merge/skulls #2: build release images for i7 model too

merge, 2018-01-16:
Basically we need to have the microcode version information. It should be tested well before releasing.

merge, 2018-01-19:
New strategy: Make sure the most recent Intel microcode updates are in coreboot upstream. For every CPU model.

We simply release when there's a new one in coreboot.

closed 2018-01-19 by merge

mikeryan, 2018-02-17:
Do you know when the microcode update supporting i7 will make it into coreboot upstream?

merge, 2018-02-19:
That will be just a few days after Intel releases it. We collect updates while we wait here: https://github.com/merge/coreboot-x230/issues/3

If we have indication that it'll take longer, I might do a release in the meantime, including the 20171117 microcode package, because for the X230, even that includes a newer microcode version than coreboot master now...

But coreboot upstream seems to wait for Intel now.

merge, 2018-02-20:
I did a release that includes the most recent microcode revision now. coreboot upstream still includes revision `1b` and the 0.0.3 image includes revision `1c` which is part of the latest release by Intel 20171117 (and was also in 20180108).

You cannot trivially rebuild that, as I added it to my coreboot tree only, but as soon as Intel finally does a new release, we'll help pushing it into coreboot and we're back on track. And I have the impression that we'll get a new revision for the X230 with that anyways.

mikeryan, 2018-02-23:
Does this mean 0.0.3 will boot on an i7 X230?

merge, 2018-02-23:
Yes, it does.

Am 23. Februar 2018 01:44:33 MEZ schrieb Mike Ryan <notifications@github.com>:
>Does this mean 0.0.3 will boot on an i7 X230?

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile