merge/skulls #3: Wait for Intel microcode release and include it in coreboot

merge, 2018-01-19:
Since 20180108 seems to have "reboot" issues, see https://newsroom.intel.com/news/firmware-updates-and-initial-performance-data-for-data-center-systems/ we currently only "prepare" to include a new microcode release in coreboot upstream.

As https://security-center.intel.com/advisory.aspx?intelid=INTEL-SA-00088 suggests, there could be a new release "soon". By then we hopefully have enough review upstream on *how* to include it and it all goes without much delay.

... It might not even affect the X230, since the 20180108 package, for X230's CPU model has updates from up until onyl 2015-02-26. Nevertheless: *currently* in coreboot we have an older version (from 2014), so we want to have it included.

merge, 2018-01-19:
According to https://support.lenovo.com/at/en/solutions/len-18282 Lenovo has schedules a BIOS update for X230 for 2018-02-02. We won't use it, but let's hope to have a release by Intel too by then.

merge, 2018-01-24:
Now 20180108 is officially pulled and deleted by Intel, see https://newsroom.intel.com/news/root-cause-of-reboot-issue-identified-updated-guidance-for-customers-and-partners/ and it sounds like a new version is not too far away.

anterguy, 2018-01-25:
As far as I understood does the 20180108 microcode update not effect the X230 (i.e. Ivy Bridge platform) since it's too old. Last thing I saw was that Intel is planning to patch the pre 2013 chips in the beginning of february, but with all those issues on the more recent ones I doubt they will be on time ;)

merge, 2018-01-25:
yes. in the 20180108 microcode package (not available anymore), for the X230's CPU ID (306ax) the latest update is 2015-02-26

I expect Intel to do changes for Ivy Bridge in their next release. I'm sure Lenovo is one of the (quote from 3 days ago) "Industry partners focus efforts on testing early versions of the updated solution"; and since they schedule a BIOS update for X230 for 2018-02-02, that must include exactly this update from Intel.

(unreleated here, but it'll be interesting of Intel does changes for Sandy Bridge too, which would help X220 users, even if Lenovo won't do a BIOS update)

merge, 2018-01-31:
Lenovo updated the availability to 2/9/2018. Nothing from Intel yet.

merge, 2018-02-08:
Lenovo updated the x230 BIOS update to "**Update withdrawn by Intel; Target TBD**"

merge, 2018-02-23:
Update from 2days ago: Intel keeps us up to date at https://www.intel.com/content/www/us/en/architecture-and-technology/facts-about-side-channel-analysis-and-intel-products.html now.

For the X230, the microcode update (MCU) is in status "beta", which means finished, but only available for vendors under NDA for testing.

The distribution channel seems to have changed though. There doesn't seem to be a Linux ucode data file release planned. The MCUs come as part of vendor BIOS updates, as they become stable.

Wheather this means we'll have to try cutting the MCU out of a Lenovo BIOS, to have it in coreboot, remains to be seen. But it looks like that'll be the quickest way for us.

What's also nice to see is that the Sandybridge cpus used in the X220 are equally supported by Intel still.

As the situation seems unclear for coreboot upstream, we might want to at least push in 20171117, in order to have the stable " pre-mitigation" state for now... I'll get in touch there ealy next week.

merge, 2018-03-05:
March 1st: still "Beta".
[microcode-update-guidance-01-03-2018.pdf](https://github.com/merge/coreboot-x230/files/1779909/microcode-update-guidance-01-03-2018.pdf)

merge, 2018-03-06:
March 4th: Lenovo updates the target date to `4/15/18`,well over a month from now -.-

merge, 2018-03-07:
March 6th: Intel marks our update as "Production" with the new Revision `1f`

yay, so where is it?
[microcode-update-guidance-06-03-2018.pdf](https://github.com/merge/coreboot-x230/files/1788129/microcode-update-guidance-06-03-2018.pdf)

merge, 2018-03-07:
Intel appearently did it's (non-public) release. We continue in a new thread about extracting: https://github.com/merge/coreboot-x230/issues/10

closed 2018-03-07 by merge