merge/skulls #4: Get rid of VGA BIOS binary

merge, 2018-01-19:
Is there a configuration for having Video pre-Linux (in SeaBIOS and bootloader from HDD) without the binary VGA BIOS rom extracted from the vendor image?

nemanjan00, 2018-01-29:
I did try to do this, but, with seabios, I do not get any image until linux takes over.

merge, 2018-01-29:
We can't just remove the vga bios. Native graphics init doesn't seem to work either. Did you try that? It's not all that bad having this as binary because of good coreboot integration and it seems to be identical for all X230 models. I just like to have this question open.

It'd probably be a seperate project analyzing this vga bios.

rectagonal, 2018-03-26:
Dumb question as im relatively new to coreboot. (I did my first flash w/ this repos build this weekend.) Is it possible to use native graphics int + SeaVGABios in lieu of the proprietary vendor vga rom?

merge, 2018-03-26:
I guess it's always "native graphics init" or "run a VGA Bios binary". I didn't yet try to build a VGA bios using the seabios repo, and add that as "external" in coreboot. Do you want to try this?

Also, it's been quite some time since I tried native graphics init, but last thing I heard was, to have graphics in a bootloader or payload, and also to support Windows, we need that extracted VGA bios.

rectagonal, 2018-03-26:
I wasn't sure How external it actually was. Coreboot wiki makes it sound like it's in the main tree just under a non obvious option:

> SeaBIOS can provide an option rom that implements legacy VGA BIOS compatibility for coreboot initialized GPUs. To use this feature select CONFIG_VGA_COREBOOT (in "make menuconfig" under "VGA ROM ---> VGA Hardware Type" select "coreboot linear framebuffer"). 
https://www.coreboot.org/SeaBIOS#Using_coreboot_VGA_support

I had planned on trying it unless someone told me it was a terrible idea, but it might be a few days before im ready to pop open the laptop again. I also don't know how I would test Windows compatibility. I mostly run NetBSD and sometimes dual boot Devuan.

tlaurion, 2018-04-29:
x230 doesn't need vga bios and can be initiated by native init code. Here is the [x230 coreboot building configuration ](https://github.com/osresearch/heads/blob/master/config/coreboot-x230.config) used by Heads to boot it's linux payload. Use that config and modify to use SeaBIOS.

merge, 2018-05-01:
are you sure? last time I checked I didnt see seabios and grub, and had display only when my host linux started.

Am 29. April 2018 22:14:52 MESZ schrieb tlaurion <notifications@github.com>:
>x230 doesn't need vga bios and can be initiated by native init code.
>Here is the [x230 coreboot building configuration
>](https://github.com/osresearch/heads/blob/master/config/coreboot-x230.config)
>used by Heads to boot it's linux payload. Use that config and modify to
>use SeaBIOS.

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

merge, 2018-05-02:
Thanks for the config link. Why do they have `CONFIG_VGA_BIOS_FILE="pci8086,0166.rom"` in their config? Is that a mistake?

tlaurion, 2018-05-05:
You can comment it out, since not used:
```
# CONFIG_VGA_BIOS is not set
CONFIG_MAINBOARD_HAS_NATIVE_VGA_INIT=y
# CONFIG_MAINBOARD_FORCE_NATIVE_VGA_INIT is not set
CONFIG_MAINBOARD_DO_NATIVE_VGA_INIT=y
```

have you tested?

Thrilleratplay, 2018-05-05:
I can confirm.  I just flashed coreboot using the last source and compiled libgfxinit.  Here is a [gist of the make file](https://gist.github.com/Thrilleratplay/be688ceafffa04da12d24654906027b1)

nemanjan00, 2018-05-05:
Is this config based off the one in repo? 

There are quite a bit of changes...

Thrilleratplay, 2018-05-05:
@nemanjan00 It is.  [Other than adding in additional payloads, the changes are related to video related](https://github.com/Thrilleratplay/skulls/commit/908249efbdb5fe2a14e0af572175616666ddab06)

nemanjan00, 2018-05-05:
Yeah, works!

merge, 2018-05-07:
Thanks for testing this. We'll still use the blobs repo for microcode updates, but I'll test this soon and hope to be able to remove the vga file. great.

merge, 2018-05-07:
Based on what @Thrilleratplay suggests I have the following config diff to the current one in our repo:

```
--- a/x230/92263853ad.config
+++ b/x230/1ecec5f979.config
@@ -76,6 +76,7 @@ CONFIG_VENDOR_LENOVO=y
 # CONFIG_VENDOR_LOWRISC is not set
 # CONFIG_VENDOR_MSI is not set
 # CONFIG_VENDOR_NVIDIA is not set
+# CONFIG_VENDOR_OCP is not set
 # CONFIG_VENDOR_PACKARDBELL is not set
 # CONFIG_VENDOR_PCENGINES is not set
 # CONFIG_VENDOR_PURISM is not set
@@ -84,6 +85,7 @@ CONFIG_VENDOR_LENOVO=y
 # CONFIG_VENDOR_SAPPHIRE is not set
 # CONFIG_VENDOR_SCALEWAY is not set
 # CONFIG_VENDOR_SIEMENS is not set
+# CONFIG_VENDOR_SIFIVE is not set
 # CONFIG_VENDOR_SUNW is not set
 # CONFIG_VENDOR_SUPERMICRO is not set
 # CONFIG_VENDOR_TECHNEXION is not set
@@ -104,7 +106,7 @@ CONFIG_PAYLOAD_CONFIGFILE=""
 CONFIG_VGA_BIOS_ID="8086,0166"
 # CONFIG_ONBOARD_VGA_IS_PRIMARY is not set
 CONFIG_DIMM_SPD_SIZE=256
-CONFIG_VGA_BIOS=y
+# CONFIG_VGA_BIOS is not set
 CONFIG_DCACHE_RAM_BASE=0xfefe0000
 CONFIG_DCACHE_RAM_SIZE=0x20000
 CONFIG_VGA_BIOS_FILE="pci8086,0166.rom"
@@ -185,7 +187,6 @@ CONFIG_SYSTEM_TYPE_LAPTOP=y
 # SoC
 #
 CONFIG_CPU_SPECIFIC_OPTIONS=y
-# CONFIG_S3_VGA_ROM_RUN is not set
 CONFIG_RAMTOP=0x200000
 CONFIG_HEAP_SIZE=0x4000
 CONFIG_RAMBASE=0x100000
@@ -296,6 +297,7 @@ CONFIG_USE_NATIVE_RAMINIT=y
 # CONFIG_NATIVE_RAMINIT_IGNORE_MAX_MEM_FUSES is not set
 # CONFIG_NATIVE_RAMINIT_IGNORE_XMP_MAX_DIMMS is not set
 CONFIG_SANDYBRIDGE_IVYBRIDGE_LVDS=y
+CONFIG_IF_NATIVE_VGA_INIT=y
 CONFIG_HPET_ADDRESS=0xfed00000
 CONFIG_HPET_MIN_TICKS=0x80
 CONFIG_MAX_PIRQ_LINKS=4
@@ -383,6 +385,7 @@ CONFIG_UDK_VERSION=2013
 # CONFIG_ARCH_ROMSTAGE_ARMV8_64 is not set
 # CONFIG_ARCH_RAMSTAGE_ARMV8_64 is not set
 CONFIG_ARCH_ARMV8_EXTENSION=0
+# CONFIG_ARM64_USE_ARCH_TIMER is not set
 # CONFIG_ARM64_A53_ERRATUM_843419 is not set
 # CONFIG_ARCH_MIPS is not set
 # CONFIG_ARCH_BOOTBLOCK_MIPS is not set
@@ -432,32 +435,28 @@ CONFIG_BOOTBLOCK_SIMPLE=y
 CONFIG_BOOTBLOCK_SOURCE="bootblock_simple.c"
 # CONFIG_COLLECT_TIMESTAMPS_NO_TSC is not set
 CONFIG_COLLECT_TIMESTAMPS_TSC=y
+# CONFIG_PAGING_IN_CACHE_AS_RAM is not set
+# CONFIG_IDT_IN_EVERY_STAGE is not set
 
 #
 # Devices
 #
 CONFIG_HAVE_VGA_TEXT_FRAMEBUFFER=y
+CONFIG_HAVE_LINEAR_FRAMEBUFFER=y
 CONFIG_MAINBOARD_HAS_NATIVE_VGA_INIT=y
 # CONFIG_MAINBOARD_FORCE_NATIVE_VGA_INIT is not set
 CONFIG_MAINBOARD_HAS_LIBGFXINIT=y
-# CONFIG_MAINBOARD_DO_NATIVE_VGA_INIT is not set
+CONFIG_MAINBOARD_DO_NATIVE_VGA_INIT=y
 # CONFIG_MAINBOARD_USE_LIBGFXINIT is not set
-CONFIG_VGA_ROM_RUN=y
+# CONFIG_VGA_ROM_RUN is not set
 # CONFIG_NO_GFX_INIT is not set
-# CONFIG_ALWAYS_LOAD_OPROM is not set
-# CONFIG_ON_DEVICE_ROM_LOAD is not set
-# CONFIG_PCI_OPTION_ROM_RUN_REALMODE is not set
-CONFIG_PCI_OPTION_ROM_RUN_YABEL=y
-# CONFIG_YABEL_PCI_ACCESS_OTHER_DEVICES is not set
-CONFIG_YABEL_VIRTMEM_LOCATION=0x1000000
-# CONFIG_YABEL_DIRECTHW is not set
 # CONFIG_MULTIPLE_VGA_ADAPTERS is not set
 
 #
 # Display
 #
-# CONFIG_FRAMEBUFFER_SET_VESA_MODE is not set
 CONFIG_VGA_TEXT_FRAMEBUFFER=y
+# CONFIG_GENERIC_LINEAR_FRAMEBUFFER is not set
 # CONFIG_SMBUS_HAS_AUX_CHANNELS is not set
 CONFIG_PCI=y
 CONFIG_MMCONF_SUPPORT=y
@@ -531,7 +530,7 @@ CONFIG_SMBIOS_PROVIDED_BY_MOBO=y
 # CONFIG_MAINBOARD_HAS_I2C_TPM_CR50 is not set
 # CONFIG_PLATFORM_USES_FSP2_0 is not set
 # CONFIG_INTEL_DDI is not set
-# CONFIG_INTEL_EDID is not set
+CONFIG_INTEL_EDID=y
 CONFIG_INTEL_INT15=y
 CONFIG_INTEL_GMA_ACPI=y
 # CONFIG_INTEL_GMA_SSC_ALTERNATE_REF is not set
@@ -660,7 +659,9 @@ CONFIG_SEABIOS_STABLE=y
 # CONFIG_SEABIOS_MASTER is not set
 # CONFIG_SEABIOS_REVISION is not set
 # CONFIG_SEABIOS_THREAD_OPTIONROMS is not set
+CONFIG_SEABIOS_VGA_COREBOOT=y
 CONFIG_SEABIOS_BOOTORDER_FILE=""
+CONFIG_PAYLOAD_VGABIOS_FILE="payloads/external/SeaBIOS/seabios/out/vgabios.bin"
 CONFIG_SEABIOS_DEBUG_LEVEL=-1
 
 #
@@ -695,12 +696,10 @@ CONFIG_HAVE_DEBUG_SMBUS=y
 # CONFIG_DEBUG_SMM_RELOCATION is not set
 # CONFIG_DEBUG_MALLOC is not set
 # CONFIG_DEBUG_ACPI is not set
-# CONFIG_X86EMU_DEBUG is not set
 # CONFIG_DEBUG_SPI_FLASH is not set
 # CONFIG_TRACE is not set
 # CONFIG_DEBUG_BOOT_STATE is not set
 # CONFIG_DEBUG_ADA_CODE is not set
-CONFIG_NO_EDID_FILL_FB=y
 # CONFIG_ENABLE_APIC_EXT_ID is not set
 CONFIG_WARNINGS_ARE_ERRORS=y
 # CONFIG_POWER_BUTTON_DEFAULT_ENABLE is not set
```

I don't have display in GRUB. SeaBIOS works though.

nemanjan00, 2018-05-07:
I do have display in GRUB. 

This is my config: 

.config: 
https://gist.github.com/nemanjan00/14127cbaef7ed5c3e6b7410825526982

grub.cfg: 
https://gist.github.com/nemanjan00/5ea1cf9e2e0bb4bcb33d1d25b758c299

merge, 2018-05-07:
you use GRUB as coreboot payload. You have display there, just like SeaBIOS has, using native vga init. What I'm talking about is GRUB on disk (or Windows on disk, which wouldn't work either, I think).

Can you confirm having video in GRUB on disk?

nemanjan00, 2018-05-07:
I am 80% sure I did. I will try it again later.

merge, 2018-05-17:
FWIW, the Video BIOS is owned by Intel, and they have some information on it
* https://www.intel.com/content/www/us/en/support/articles/000005749/graphics-drivers.html
* https://www.intel.com/content/www/us/en/intelligent-systems/intel-embedded-graphics-drivers/faq-bios-firmware.html

but in order to improve the situation, we'd have to get a statement from Intel about being allowed to distribute this.

Thrilleratplay, 2018-05-17:
I do as well using GRUB but I have only booted into Linux.  Based on the how the Coreboot wikis for the x220 and x230 are worded, Windows or Windows bootloader may still require the VGA binary.

merge, 2018-05-18:
Please post your coreboot config file. I've never found a configuration that worked, but I'd love to find one.

nemanjan00, 2018-05-18:
https://gist.github.com/nemanjan00/b7e63adc48514f13f8f536c855757b91

merge, 2018-05-20:
I can confirm that this does not work for me. I don't see my GRUB menu with that config -.- what could I am be doing wrong? Did you install GRUB after flashing maybe?

Thrilleratplay, 2018-05-20:
@merge  If you using SeaBIOS, do you see the SeaBIOS output but not the GRUB menu? 

I mostly use GRUB to handle the disk encryption and hide the menu.  Use the basic console but did not disable the graphical console.
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet loglevel=1 vga=current"

# Uncomment to use basic console
GRUB_TERMINAL_INPUT=console

# Uncomment to disable graphical terminal
#GRUB_TERMINAL_OUTPUT=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command `vbeinfo'
GRUB_GFXMODE=auto

# Uncomment to allow the kernel use the same resolution used by grub
GRUB_GFXPAYLOAD_LINUX=keep

# Uncomment one of them for the gfx desired, a image background or a gfxtheme
#GRUB_BACKGROUND="/path/to/wallpaper"
#GRUB_THEME="/path/to/gfxtheme"
```

These are the relevant lines from my grub config.  If this does not provide enough information, I can post the whole thing in a gist.

@nemanjan00 Are you using console output or graphical output for GRUB?

merge, 2018-05-21:
I see seabios, but not GRUB, yes. I use Debian's default installation, which is what I always want to have supported, so trying to change grub configs doesn't really help for now.

nemanjan00, 2018-05-21:
Part of my grub.cfg I think you are asking about: 

```
terminal_input console
terminal_output gfxterm
```

merge, 2018-05-22:
when configuring a linear "high resolution" framebuffer in coreboot, even the graphical GRUB console works. I don't see Linux' boot log afterwards though, so that's not yet a possible config. But still, we should find a way to add a text-based image, with a dependency on `GRUB_TERMINAL=console`.

merge, 2018-05-23:
I simply don't see early kernel log messages, when using `CONFIG_GENERIC_LINEAR_FRAMEBUFFER` (which you don't have enabled in your posted config). Without this (in text mode), all is fine.

merge, 2018-11-23:
going forward, there's #46 with the remaining problem I see

closed 2018-11-23 by merge