merge/skulls #6: Settings payload

nemanjan00, 2018-01-25:
https://www.reddit.com/r/thinkpad/comments/68ez34/thinkpad_x230_coreboot_button_swap/dgxy45b/

nemanjan00, 2018-01-25:
I had to clear CMOS after these changes. 

Just enable CMOS settings loading and we can set settings using nvramtool...

merge, 2018-01-25:
what config option exactly do you mean to enable?

nemanjan00, 2018-01-25:
Load settings from CMOS, or something like that, I can not remmember. Will
take a look when I get home

nemanjan00, 2018-01-25:
Inside general: 
[image]

merge, 2018-01-29:
Thanks for your help! I plan to do a THANKS file. If you want that, give me your name and email address, and I'd be happy to include you. so, thanks.

closed 2018-01-29 by merge

nemanjan00, 2018-01-29:
Thank you for making this happen. :)

I do not think I could configure my build environment without your config. 

BTW, how about making IRC channel so we can discuss what improvements we can make? 

I have a few ideas...

merge, 2018-01-29:
I think setting anything more than a repo up would be overkill. Remember we simply use coreboot here, and even that only for creating a valid image for 4MB of the 12.

If you have an idea, why not open an issue here or post it on the coreboot list?

commit 2341a9b0 (2018-01-29): update config / add nvramcui CMOS settings payload

This resizes CBFS to be 4M and includes the nvramcui payload, see
https://www.coreboot.org/Payloads#nvramcui

Closes #5
Closes #6