merge/skulls #9: Windows compatibility?

tjburrows, 2018-03-04:
I flashed 0.0.3 to my x230 and Windows, even in safe mode, hangs and won't boot.  After flashing do I need to reinstall windows?  What is the expected behavior?

merge, 2018-03-04:
I haven't tested Windows in a while. Can you access the seabios boot menu normally? Basically seabios supports running Windows. Maybe you find details on their project pages. What I did once was a fresh installation of Win 7...

Am 4. März 2018 19:59:49 MEZ schrieb Travis Burrows <notifications@github.com>:
>I flashed 0.0.3 to my x230 and Windows, even in safe mode, hangs and
>won't boot.  After flashing do I need to reinstall windows?  What is
>the expected behavior?

-- 
Martin Kepplinger
sent from mobile

nemanjan00, 2018-03-04:
Is it UEFI? 
I remmember having some problems with even UEFI linux. (I am not sure if I switched from UEFI since then)

tjburrows, 2018-03-04:
I have grub2 installed with debian and windows dual booted.  Seabios sends me to the grub menu, and then I can select OS from there.  Debian worked fine, but seemed to have some issues with the display, flickering and adjusting the display a bit on startup.

tjburrows, 2018-03-06:
So I reflashed the Lenovo BIOS, and set boot to Legacy boot, but it was on 'Both', which I doubt would cause an issue.  I reflashed 0.0.3, and it behaved exactly the same way as before.  I suppose the next step would be to try reinstalling Windows and seeing if it works, rather than trying to use an existing install.

merge, 2018-03-26:
Did anybody do a fresh Windows 7 installation on our (latest) release (or similar build)?

tjburrows, 2018-03-27:
I have not yet tried a fresh reinstall.  I did try the latest build and I got the same results as before.  One of these days I will do the reinstall.

schmatzler, 2018-04-08:
Windows 10 works fine with 0.0.4 on my X230T.

It detected a few different devices at startup and that's it. I can even use touch + pen without any problems.

tjburrows, 2018-04-13:
Windows 10 works for me too!

Kaito-RTRT, 2018-04-17:
Windows10+MBR SSD+GRUB+Archlinux works pretty well here. However, Windows10 boots up relatively slow.

merge, 2018-04-17:
ok. since end of (even extended) support for windows 7 is 1,5 years away already, I guess only Windows 10 is relevant anyways.

Thanks _very_ much for all the feedback! This is a milestone towards a 1.0 release. :+1:

schmatzler, 2018-04-17:
Windows 10 actually boots very fast on my machine, if I disable the touchpad and only use the trackpoint.

But there are two bugs present:
1. It doesn't booth with an eGPU attached. ACPI_BIOS_ERROR bluescreen
2. The machine is much more sensitive to static electricity now. This may sound weird, but I often get little static shocks in my office when touching the device. Must be because of the dry air.

I never experience problems with the stock BIOS, but with coreboot, it very often locks up when I touch the device. This makes it pretty much unuseable for me.

merge, 2018-04-17:
ok. thanks for reporting! no 1.0 just yet :)

I somewhat have to rely on you with Windows testing. Maybe we should document how to report these bugs upstream...

Am 17. April 2018 18:47:25 MESZ schrieb Ronny Schmatzler <notifications@github.com>:
> Windows 10 actually boots very fast on my machine, if I disable the
> touchpad and only use the trackpoint.
>
> But there are two bugs present:
> 1. It doesn't booth with an eGPU attached. ACPI_BIOS_ERROR [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

gazny, 2018-08-06:
When ever I boot into w10 all my bios setting are reset and my time is wrong. It works fine in Linux but not in w10, I will test the new release 0.9 to test if this is still an issue

Thrilleratplay, 2018-08-06:
@gazny Other than the time, what other settings are reset?   If it is just the time, your Linux distro likely   runs ntp when first connected to the internet.  Sync your system clock with your hardware clock in Linux and enable ntp in Windows.  Various encryption algorithms break (ex SSL) if the time is too far out of sync.

gazny, 2018-08-07:
@Thrilleratplay FN/ctrl swap, usb on, touch pad off, debug level would be on spew. and boot mode was set to backup
Settings are no longer reset when booting into windows. but s3 still pauses the time on windows.

khseal, 2018-11-06:
It's bad that win7 does not work = (

merge, 2018-11-07:
It is. It's purely a coreboot topic though...

khseal, 2018-11-07:
> It is. It's purely a coreboot topic though...
I read the documentation of coreboot.
I found this
Windows 7: get stuck at disk.sys,classpnp.sys

taury55, 2019-08-02:
I just flashed my X230 laptop which have different memory chips (bottom: "W25Q64.V" if you want to add to compatibility table and top one I can't remember...).

I installed fresh Windows 10 and it works smoothly except I can't get to working Trackpoint, Touchpad and buttons for scroll left and right click. I know I must force install driver for that but it doesn't detect them.
If someone knows how to make it work please say.

But thank you for free BIOS.

merge, 2019-08-22:
> I just flashed my X230 laptop which have different memory chips (bottom: "W25Q64.V" if you want to add to compatibility table and top one I can't remember...).

thanks for reporting. would be good to add if you would have the full output from flashrom for both chips.

> 
> I installed fresh Windows 10 and it works smoothly except I can't get to working Trackpoint, Touchpad and buttons for scroll left and right click. I know I must force install driver for that but it doesn't detect them.
> If someone knows how to make it work please say.

the only thing that comes to mind is that you might have it disabled somehow using a fn-key, if that's possible at all?

> 
> But thank you for free BIOS.

tjburrows, 2020-01-06:
I would like to report that I tried flashing the free seavgabios version of skulls 0.1.8 and I can confirm that Windows 10 does not boot.  It was worth a try.  The W10 boot logo comes up and the wheel spins for a while, and then the computer restarts.  The W10 bootsplash is squeezed to the left half of my laptop screen, using half of the width and the whole height of the display.

late-in-autumn, 2020-03-13:
I have communicated with other Coreboot users, and they have all reported that SeaVGABIOS does not work with Windows for now. I have flashed the "non-free" build and Windows 10 mostly works, except that I have discovered an issue related to S3 suspend and RTC/system clock: issue #146.

late-in-autumn, 2020-03-13:
A side note is that after flashing the Skulls firmware, an unknown PCI device may pop up under Windows (at least with my X230T), and it seems to be the Intel DPTF interface by Googling the PCI ID. This can be worked around by installing the DPTF driver from later ThinkPad models. I have tried the ones from P72 and T480 and they work fine. I suspect that this interface was not exposed by the Lenovo firmware, but by Skulls, and that's why Lenovo did not provide a DPTF driver for the X230T.

divad42, 2021-05-07:
Intel's graphics driver crashes my Windows 10 install on my X230.  It's stuck in SeaBIOS's resolution, which also happens to be wrong (cut off on the sides).

MichNLGithub, 2022-10-28:
> Intel's graphics driver crashes my Windows 10 install on my X230.  It's stuck in SeaBIOS's resolution, which also happens to be wrong (cut off on the sides).

Same problem with the installation of the Intel graphics 4000 driver.