merge/skulls #12: Offer: Installation service

merge, 2018-03-26:
If you want to have our documented setup on your X230 and cannot do it yourself, you shouldn't be totally out of luck.

I offer to do the whole flashing procedure for you, flashing the latest release we have here, and applying me_cleaner and all that fun, as I have time.

You'd need send the machine in to me and cover shipping costs (and optionally add a tip) via BTC `1EH4Y37Zgves8YHJybLA9mw5aaqS7BtLs7` or LTC `LeNrCxfdZ3qJMjTjjKU4z4brDfNAjrpBWE`. Details by email: martink@posteo.de

Also, sometimes I might have an X230 available to sell. Ask if you're interested.

nemanjan00, 2018-03-26:
Hello, 

For anyone in Belgrade or Šabac, feel free to contact me and I will help you flash coreboot on your X230.

abhas, 2018-03-29:
I thought I could add myself to this "list". I am based in Bangalore, India and in case anyone requires help with Coreboot installation on X200 / X220 / X230 or a unit with Coreboot pre-flashed, I would be willing to help you out.

tlaurion, 2018-04-29:
Offering similar services and starting an enterprise based in Canada to offer similar services + training to change numeric hygiene and make Heads+QubesOS available to less Techsavvy people.

Am a QubesOS/Heads contributor,[ ported Heads to the KGPE-D16 server platform](https://github.com/osresearch/heads/issues/134), and looking to make Heads more mainstream. Merging [user signing keys into Heads is becoming streamlined](https://github.com/osresearch/heads/pull/357). [Salt deployment formula](https://github.com/SkypLabs/my-qubes-os-formula/) are also a thing to easy, utilately permitting to inject salt configurations after QubesOS installation and prior to system configuration.

I am looking to co-create a cooperative to facilitate trustworthy hardware acquisition at lowest cost possible (Nitrokey Pro, x230, KGPE-D16), since price for hardware is always determined by buying volumes. Contact me if interested.

Started negotiation with Vikings.net (KGPE-D16), Nitrokey and Neweggs Business.

Feel free to contact me at insurgo@riseup.net.

merge, 2018-05-17:
@tlaurion thanks for the offer, feel free to ask me if you're ordering x230 laptops.

While I'll leave this issue open for now, I want to make it so obvious and simple that it won't be needed by most people. Maybe I'll close it when

- [ ] #26 works reliably and all traces of chipnames are buried in the scripts
- [ ] #24 offers anyone reliable sources for flashing hardware
- [ ] we have the ch341a usage extensively documented and supported, just like the RPi
- [ ] #23 is done and we can build reproducibly (maybe)

merge, 2018-07-30:
I'll go ahead and close this. an installation offer, if desired, should be placed elsewhere than in a github issue. Let's focus on supporting more laptops and flashing hardware here :)

closed 2018-07-30 by merge