merge/skulls #13: Additional payload problems

nemanjan00, 2018-03-26:
I just used your config and latest version of coreboot and after adding additional payloads (tint, nvramcui, etc.), I am unable to start them. 

Laptop just reboots. 

I will try to flash your image right now, just in case it is problem with my build enviroment.

merge, 2018-03-26:
You might simply run out of space with the 4MB?

Theoretically I think we could make use of most of the 8M chip for bios too: after applying me_cleaner and re-arrange the partitions; pull extracted blobs into the coreboot build, create a 12M image (and possibly split that up again for hardware-flashing). But the result would be

* too hardware-dependent (I think; at least for now)
* inconvenient to handle for hardware-flashing
* force people to have me_cleaner applied

That said, at least I'd be very much interested at seeing such a working configuration.

nemanjan00, 2018-03-26:
Doesn't coreboot build script report if it wuns out of space?

nemanjan00, 2018-03-26:
After recompiling with just nvramcui payload, it still does not work...

merge, 2018-04-02:
Do you have any open problem here? Does building our configuration fail for you? thanks

nemanjan00, 2018-04-02:
It doesn't fail, but, nvramcui does not work in my build with your config.
For which Arch are you building it?

merge, 2018-04-02:
i386

merge, 2018-04-02:
So building this should be possible. I don't do anything special, see https://www.coreboot.org/Build_HOWTO

* have the repo and submodules checked out
* `make crossgcc-i386`
* have the config correct (either checkout the state I have or update your config to your current state)
* have the VGA bios file in coreboot's top level dir
* `make`

just for reference. I know others have built based on this configuration already.

nemanjan00, 2018-04-02:
Changing x64 to i386 helped resolve this problem...

closed 2018-04-02 by nemanjan00