merge/skulls #16: Confirm integration of microcode into flashed bios?

rennis250, 2018-03-27:
I've successfully flashed coreboot onto my x230 and it boots without problems, but I wasn't sure how to tell if the microcode from the Windows/Lenovo updates was properly integrated into the coreboot-ed bios. Is there a standard way to ensure that it was copied? How do I confirm this?

Thanks!

merge, 2018-03-27:
If you're running Linux, run `dmesg | grep microcode` and verify you're running revision `1f`. Also, you can run something like https://github.com/speed47/spectre-meltdown-checker and see that you have the instructions they added for this purpose.

If you're running Windows, I'm really not sure, but there must be a similar tool to check the "hardware" state with respect to spectre/meltdown mitigations.

rennis250, 2018-03-27:
Ok, I get revision ```1b```. I guess I need to include the binary outputs of ich_descriptors_tool in the bios that coreboot builds to get newer then? I've been doing all of this myself to understand the process and for fun, so to be clear, I'm not using the files provided on your distribution here (although, I think it is very kind of you to provide them! :) ).

rennis250, 2018-03-27:
Also, did you ever run into the output from intelmetool like what I found here: https://github.com/corna/me_cleaner/issues/184

My initial post on that issue can be ignored. The relevant output is in the other posts.

merge, 2018-03-27:
I see. Yea that's not documented well enough maybe. coreboot upstream still has `1b`. For the current release, we add Intel's new package and have the change up for review upstream here: https://review.coreboot.org/#/c/blobs/+/23315/

You should be able to cherry-pick it inside of the blobs repository (submodule in coreboots repo).

Is that clear?

me_cleaner is really a different topic, but I'll run intelmetool on my system which has applied what we have documented here, soon.

rennis250, 2018-03-27:
Yup, that's clear. Thanks! I'll give that a try and looking forward to what you find with intelmetool.

merge, 2018-03-27:
```
# ./intelmetool --me
Bad news, you have a `QM77 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI found: [8086:1e3a] 7 Series/C216 Chipset Family MEI Controller #1

ME Status   : 0x1e020191
ME Status 2 : 0x100a0140

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Initializing
ME: Current Operation State : Bring up
ME: Current Operation Mode  : Debug
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Check to see if straps say ME DISABLED

ME: Extend SHA-256: 099b87484854301020c4c8cb95854b0c080ee7acf6d701104e20c7a3d7583410

ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready

```

rennis250, 2018-03-27:
Thanks! According to https://github.com/corna/me_cleaner/issues/184#issuecomment-376483387, this is the correct output for a disabled ME when one has also used the "-S" flag to me_cleaner.

closed 2018-03-27 by rennis250

rennis250, 2018-03-27:
Oh, and I forgot to confirm that applying your latest changes to the 3rdparty blobs works. I now get "revision=0x1f" from the microcode line of dmesg. Thanks!

merge, 2018-03-27:
thanks for the feedback