merge/skulls #17: Cannot flash on X230 Tablet (i7)

schmatzler, 2018-04-08:
It seems that my X230 Tablet model (Ivy Bridge) actually has two 8MB flash chips instead of 4MB+8MB.

I have a CH341A USB flasher and it reads the two chips as follows.

Upper one:
    flashrom --programmer ch341a_spi
    Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.

Bottom one:
    flashrom --programmer ch341a_spi
    Found Macronix flash chip "MX25L6405" (8192 kB, SPI) on ch341a_spi.
    Found Macronix flash chip "MX25L6405D" (8192 kB, SPI) on ch341a_spi.
    Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.
    Found Macronix flash chip "MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E" (8192 kB, SPI) on ch341a_spi.
    Multiple flash chip definitions match the detected chip(s): "MX25L6405", "MX25L6405D", "MX25L6406E/MX25L6408E", "MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E"
    Please specify which chip definition to use with the -c <chipname> option.

I cannot flash the coreboot toprom, because it's only 4MB. Am I doing something wrong here?

schmatzler, 2018-04-08:
Fixed the problem.

I am not sure what the problem was, but I have the two chips MX25L3206E and MX25L6406E on my board. Flashrom just didn't detect them properly at first.

I cleaned up both chips with isopropyl and suddenly the detection worked fine.

I also needed to modify my CH341A to supply 3.3V and not fry the chips using these instructions:
https://www.eevblog.com/forum/repair/ch341a-serial-memory-programmer-power-supply-fix/

Thanks for providing these images!

closed 2018-04-08 by schmatzler

merge, 2018-04-08:
good to hear. thanks for the feedback

Am 8. April 2018 18:02:57 MESZ schrieb Ronny Schmatzler <notifications@github.com>:
> Fixed the problem.
>
> I am not sure what the problem was, but I have the two chips MX25L3206E
> and MX25L6406E on my board. Flashrom just didn't detect them properly
> at first.
>
> I cleaned up both chips [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile