merge/skulls #18: internel flashing 4MB only / release management

merge, 2018-04-13:
first: we don't (yet) want to advertise internal flashing. It still feels like it could be dangerous. BUT:

In case you _do_, how do you do it? Is there a way for flashrom to take the 4MB "top" image and write it _to_ address `0x00800000` (from `0x00800000` to `0x00bfffff`) instead of starting at `0x00000000`?

If that _cannot_ be done: We should actually create a script that:
* creates a working dir
* creates a 12MB image out of the 4MB release image (filling with zeroes)
* creates the flashrom layout file
* prints everything necessary to know (the commands) how to flash it

The thing is, I want to stop distributing the 12MB image. It should really be generated by users (the bottom 8MB are useless).

As a sidenote, I want to think about how to do releases. Strictly speaking we should include Intel's microcode license. And if we add a script for internal flashing, described above, we should probably create a tarball.

merge, 2018-04-13:
I wrote a first draft of such a script. It's completely untested! Please, if you can, take a look at it and test it. I'll do so too, sometime "soon". thanks a lot!

nemanjan00, 2018-04-13:
I actually do use internal flashing and it proved itself very safe. And
yes, you can flash 4mb flash separately...

wrote:

I wrote a first draft of such a script. It's completely untested! Please,
if you can, take a look at it and test it. I'll do so too, sometime "soon".
thanks a lot!

merge, 2018-04-13:
thanks for the feedback. how do you currently do it?

merge, 2018-04-13:
thanks. that's what I meant :) The new "prepare" script takes the (released) 4MB image and creates a 12MB one, along with the layout-file and instructions on how to flash it internally.

Could you have a look at it? That's be of help.

nemanjan00, 2018-04-13:
I just woke up so, I am mostly confused on what is happening... 

I will take a look at it now...

nemanjan00, 2018-04-13:
Looks great to me. 

I am thinking that actually it might not be that bad idea to give option to read 8MB image instead of generating it with dd... 

So it can later be extended with more features (for example ME removing and using additional space)

merge, 2018-04-13:
well. yes. we could do a lot :)

I actually don't want to run flashrom from _this_ "prepare" script. This should be runnable by users, not only by root.

Also I want to be careful here, because it's required to have the 8MB chip flashed externally at least once (for unlocking), before being able to flash internally. Can we detect that when trying to flashrom-read internally? Or does reading already fail if not unlocked?

But you're right...
* we could add an option to provide a full or 8mb (already read) file, _that_ we update instead of prepending zeroes (only do that by default)
* we could create _another_ script that really runs flashrom, and _uses_ the above script. I clearly have to test flashing internally myself. We'd need something that works for everybody. But at least we could come up with something people can test...

In a first step, I'd like to have the current script tested and ready. I updated the README a bit. Future releases won't include a 12MB image.

nemanjan00, 2018-04-13:
I am thinking about creating testing platform for this. I have board i do
not use. IIt might be interesting to connect it and programmer connected to
it online so we can use auto testing... Also, I would connect video capture
card to internet...

merge, 2018-04-13:
I guess we'll always require to write an unlocked image once, using a hardware flasher. But could we use https://github.com/google/pawn to initially _get_ the image so that we could _only_ write the unlocked one once.

That's save quite some time. Anyone cares to compare original bios images read via hardware with the results from this tool?

merge, 2018-04-13:
reading is not possible either? I created a new issue #19 for this question.

I tested internal flashing using the script now. seems to be fine... I'll leave this issue open for a while in case we get another test.

nemanjan00, 2018-04-13:
I will try reading Bios on T430, when I get home. I do not have locked x230
image...

merge, 2018-04-13:
with the 12mb-generating "prepare"-script tested, and a release script added now, this issue is actually resolved. #19 is more interesting anyways :)

closed 2018-04-13 by merge