merge/skulls #21: create a logo for the project

merge, 2018-04-21:
in case you noticed, I renamed the project. If you feel like it, please post a suggestion for a logo. It can be a hand-drawn draft that can later be put into a vector-graphics file. My drawing skills are rather limited. Oh. and if it's a skull, it should be a friendly one :)

background: Basically I renamed and refactored the project to make room for more. We plan to add more hardware flashers to the x230 docs/scripts and the README became quite large already. Now new separate device-independent stuff out.

Also we have room for any other device now, in case you have one and think that flashing could be simplified. I certainly don't plan to myself, but why not.

Feedback and changes are always welcome. thanks so far.

nemanjan00, 2018-04-21:
I was just wondering what the goal of project like this would be... 

What do you want to do with it that coreboot does not do? 

I saw this repository as coreboot config and flashing documentation....

merge, 2018-04-21:
Initially I really just wanted to save my config. It turned out to be more fun than expected. Now it's (almost) a coreboot distribution, just like libreboot or HEADS.

goals (if it's not obvious, our docs could probably improve):
* create the most basic and easy to use coreboot system (so that people don't _stop_ using it because it's complicated to use). We use seabios, libreboot uses GRUB, HEADS uses, well, HEADS, but I think you can build Linuxboot or others with it too
* make it easy to install
* be a tester for upstream

Maybe it's as a starter-drug that encourages people to use HEADS later.

nemanjan00, 2018-04-23:
[image]
Just had some spare time and implemented my idea. :D

nemanjan00, 2018-04-23:
Just had some free time and implemented my idea... 
[image]

merge, 2018-04-23:
thanks :) I don't consider it really friendly to look at, though. Also, the project isn't called skullboot. It's skulls. I appreciate the effort though, and nice drawing skills!

Am 23. April 2018 16:19:21 MESZ schrieb "Nemanja Nedeljković" <notifications@github.com>:
>Just had some free time and implemented my idea... 
>[image]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

nemanjan00, 2018-04-23:
I actually stole skull, just added ThinkPad red and text

merge, 2018-04-24:
ok, we can't steal anything and should probably remove the image from here too.

nemanjan00, 2018-04-24:
It is free to use with modification... Even for commercial use

Thrilleratplay, 2018-05-08:
[image]

I know, It looks like Munch's The Scream and still not friendly but wanted to suggest ideas like the SOIC8 mouth or Coreboot logo used for the sinus cavity.  As inspiration, what about a [logo similar to that of Jinx hackware](https://en.wikipedia.org/wiki/Jinx_(clothing))

apraile, 2018-11-22:
(sorry for my English)

Hi,

Since the [coreboot logo](https://www.coreboot.org/assets/images/banner.svg) is a rabbit, why not a rabbit skull logo for the project? Although this is the first time with a vector graphics editor, I have created the following logo that has a Thinkpad's red cap :smile: (in SVG format [here](https://gist.github.com/apraile/e124d4ab9b959f8d30d13959a28c6fab)):

[image]

With more design knowledge the logo could be improved (shadows, 3D effect...). There are many more [examples](https://duckduckgo.com/?q=rabbit+skull&t=h_&iax=images&ia=images&iaf=type%3Aphoto-clipart) on the internet.

merge, 2018-11-23:
I like the rabbit idea. Would you change it a bit? And I'd rather have it black/white only. (we don't want to be thinkpad-specific actually; plus black/white is a bit more accessible and easier to print). And maybe you could improve the ears' (or head's) shape so that it's more obvously a rabbit... but I don't know how exactly :) 

it's really nice that you drew that! thanks so much.

apraile, 2018-11-23:
Yeah, he actually looks more like an alien than a rabbit :smile:. I will try to fix it, perhaps make it narrower.

Thrilleratplay, 2018-11-23:
It reminds me of [Frank the Bunny from Donnie Darko](https://www.google.com/search?q=frank+the+bunny&tbm=isch).  Not a bad thing in my opinion..

varac, 2018-12-30:
I like most of the logos presented here! Can't we display them in the first boot stage on the screen, like the BIOS vendors do/did ? That would increase the visibilitz of this project/coreboot dramatically.

varac, 2019-01-08:
never done it myself but there's this: https://www.coreboot.org/SeaBIOS#Adding_a_graphical_.22bootsplash.22_image

apraile, 2019-01-11:
Thanks to a [post](https://www.reddit.com/r/thinkpad/comments/9i7kmj/finally_got_my_seabios_bootsplash_to_work_on_x230/) on reddit, I created a bootsplash [JPG](https://github.com/apraile/skulls/blob/master/x230/media/bootsplash.jpg) image to test . And another version in [XCF](https://github.com/apraile/skulls/raw/master/x230/media/bootsplash.xcf) to edit it in GIMP (after editing, you have to export it to 1024x768 JPG as explained in the post).

Also, there is [`BOOTSPLASH_IMAGE`](https://coreboot.org/status/kconfig-options.html) coreboot compile option that may be easier to integrate into the project than with `cbfstool`.

merge, 2019-01-14:
I hope I find time to test... Thanks a lot for the graphics! I kind of like it and would happily use that as the logo.

So we'll add the jpg that coreboot needs for building, and update the photo on the main README page :) cool

varac, 2019-01-14:
@apraile Great, this looks really nice! Can't wait to have it on my device !

 I'd propose to (somewhere/somehow) add the term "coreboot" to it so ppl recognized the underlying foundation of skulls and to give credits to the great work of that project. What do you think ?
"skulls + coreboot" also makes searching for the skulls project a lot easier, for ppl passing by and wondering what it is.

varac, 2019-01-14:
Maybe sth like:

```
          SKULLS
   based on coreboot
press esc for boot menu
```

?

merge, 2019-01-14:
so I tested this a bit and it works right away for the nonfree config. it looks _awesome_ and I pushed the changes so you should be able to rebuild.

it seems like running an option rom is required, and I have _not_ yet gotten it to work for the seavgabios version. Who knows more?

So, I added the mentioned image to the repo now, "temporarily", but I'd also like "coreboot" or "coreboot.org" to be in there somehow, so I'd happily change it (or you just do a PR).

In case we cannot get this to work for the free version, I think we should only add it to the nonfree version and do a release anyways; we can open a new issue for that.

apraile, 2019-01-14:
Thanks @merge for the work! A pity that does not work with the seavgabios version (I use the nonfree version and I have no experience with the seavgabios version).

I agree to make a mention to coreboot, I leave it in your hands to make the changes.

merge, 2019-01-14:
XCF version 11 is not supported by my GIMP 2.8 -.- could anybody change the text to

```
        Skulls
  https://coreboot.org
Press ESC for boot menu
```

that'd be great

apraile, 2019-01-15:
I have updated the above [post](https://github.com/merge/skulls/issues/21#issuecomment-453579034) with new files. I hope the [legacy XCF](https://github.com/apraile/skulls/raw/master/x230/media/bootsplash_legacy.xcf) version can be edited.

merge, 2019-01-15:
thanks!! @apraile do you want to add the XCF to our repo, or are you happy when I just grap and add stuff from you?

apraile, 2019-01-15:
> thanks!! @apraile do you want to add the XCF to our repo, or are you happy when I just grap and add stuff from you?

Yes, you can grap and add stuff, no problem.

merge, 2019-01-17:
It seems like we have a logo. As soon as someone takes a photo for the READE, it'll be more visible. For the remaining problem I see, I've created https://github.com/merge/skulls/issues/59

thanks again for all your help here!

closed 2019-01-17 by merge

varac, 2019-02-06:
@merge Can you please quickly summarize the state of this ? Will we get a skulls bunny image now using the release images ?

merge, 2019-02-06:
@varac you'll see the bunny skull logo as splashscreen image when you flash our "normal" image. the "free" one, doesn't include it.

but: the README should say it all. Directly at the top. have you read it? Should we improve it? how?

thanks.