merge/skulls #22: Disable battery validation check patch

Thrilleratplay, 2018-05-09:
I am not sure if this fits with the philosophy and goals of Skulls but what about incorporating an EC patch that disables the authentic battery validation check?  

http://zmatt.net/unlocking-my-lenovo-laptop-part-3/
https://github.com/eigenmatt/mec-tools
https://github.com/hamishcoleman/thinkpad-ec

merge, 2018-05-13:
As I understand it, Lenovo's original flash update tools have to be used. Also, skulls is a bios image; we don't touch the EC. But we mention it in our docs and could document this possibility...

merge, 2018-05-14:
thanks for mentioning this! I'd love to have an ISO image (using the latest lenovo one) with only the battery EC patch applied.

Even if we won't be allowed to distribute it, that'd be great

merge, 2018-05-21:
I think our current documentation is about as much as we can do here. I dont see a way to automate this using free software, so I'll close this. Feel free to reopen or create a new issue if what is documented can be automated. thanks a lot!

closed 2018-05-21 by merge

kaffeekanne, 2019-07-30:
@merge:
> If you're looking for an easier target: I created a "battery patched" USB/CD image from the latest Lenovo ISO image; it's documented and ideally we could offer an image to download too...

Actually i am looking exactly for that. The battery of my x230t died years ago and i wanted to do a last EC update before moving to skulls. I have no windows around to do a flash from there, but sketched out a manual way without battery check including a Windows PE ISO and the Update tool from lenovo, interrupting the update at some point (alle files extracted) and starting the flash tool with a switch ignoring the battery state.

Since that is pretty hacky and i have not found some else having done this before, a "USB/CD image" that has done this before would be highly appreciated!

merge, 2019-07-30:
are you talking about using a cheap, 3rdparty aftermarket battery, or some check during flashing that any battery is inserted at all?

This issue is about the first problem. If the lenovo flashing tool wants the battery to be "not empty", that's a different issue. But you only need an EC update of your version really is older than 1.14.

kaffeekanne, 2019-07-30:
I actually have EC version 1.14, but BIOS version 2.60 and thought i need both to be most recent to be on the save side before flashing skulls. If most recent EC but BIOS is mostly irrelevant that might be good to add in the documentation. Lenovo still releases BIOSes so someone wanting to install skulls, probably would not need to update one last time since EC version is not increasing anymore.

Also, thanks for your hard work!