merge/skulls #23: Script for building locally

tjburrows, 2018-05-11:
I think a script to build the binary ROM locally would be a great addition.  This wouldn't be too difficult, I wouldn't think - just pulling the files from skulls and coreboot repositories, and then running the compilation commands.

Thrilleratplay, 2018-05-11:
I have an semi automated build script and a docker container.  I need to clean it up a bit and want to make sure that it will create reproducible builds.

merge, 2018-05-13:
If we can build reproducibly, it would make sense and be great to include it. It wouldn't affect release tarballs though.

Thrilleratplay, 2018-05-15:
Here is the [initial commit for a local docker builder](https://github.com/Thrilleratplay/skulls/tree/master/docker_builder).  I want to add the ability to add take a commit hash or tag.

I opted for the official coreboot docker container.  While much larger than the one I had, coreboot's is designed for reproducible builds.

merge, 2018-05-15:
Have you tested it? There are minor things I spot:
* build the toolchain if not yet built (or possibly anyways, to be sure to be up to date; it doesn't take long when up to date)
* the model name "x230" could be defined more centrally (directory name), and not hardcoded at multiple places
* running make nconfig seems strange. why have something interactive in there?
* integration: copy our config and vga bios in

but when I find time to test, I can as well imagine integrating this and impove it from there. thanks. feel free to create a pull request.

Thrilleratplay, 2018-05-15:
To answer a few of these,
*  Yes, I have tested it and one of the perks of using the official Coreboot container is that it already has the toolchain built and installed.
*  (_facepalm_) Of course I should be reading directories instead of maintaining a list.
*  Currently, it pulls the latest from Coreboot; `make nconfig` was make any relevant config changes.  When adding in the ability to build from a specific commit or tag, I will add an interactive mode flag.
* I have not a conclusive answer on why the vga bios is not already included in the coreboot blob repo.  Have you found the license agreement that permits this?  This concern is why I am using the libgfxinit config but would prefer to full integrate with the rest of skulls.

Thrilleratplay, 2018-05-15:
Hiding in plain sight

* https://github.com/coreboot/coreboot/blob/master/util/docker/coreboot-sdk/Dockerfile
* https://hub.docker.com/r/coreboot/coreboot-sdk/

merge, 2018-05-15:
thanks. if I understand this, that's irrelevant as long as _our_ config produces the same hash.

merge, 2018-05-29:
Thanks! I'd love to include them when they build reproducibly.

merge, 2018-06-18:
included. @Thrilleratplay can I add a "author" or copyright comment with your name to your scripts, as I didn't preserve the git history? thanks.

closed 2018-06-18 by merge

Thrilleratplay, 2018-06-18:
@merge Sure.  Add whatever you feel is appropriate.