merge/skulls #24: links to buying flashing hardware?

merge, 2018-05-15:
As much as I like buying stuff locally, it'd still help our cause if we'd have links to shops that ship the pomona clip and jumper cables worldwide (or multiple if regional only).

Would you add such links? If yes, how and where?

If at all, I'd even add a serial cable for the RPi. (And yes, for people who don't own a RPi we should include the ch341a, but we'd have to document it's usage first.)

merge, 2018-05-16:
thanks! I started by pointing to manufacturers, as examples for the hardware _always_ required.

a "where to buy" section with vendors, and specifics to the flashers we support, is to be done.

nemanjan00, 2018-05-16:
These are the ones I used. 

Seller I got them from does not have them any more: 

https://www.ebay.com/sch/i.html?_odkw=P5003+SMD+IC&_osacat=0&_from=R40&_trksid=p2045573.m570.l1313.TR0.TRC0.H0.XP5003+test+hook.TRS0&_nkw=P5003+test+hook&_sacat=0

Also, I used ch341a

commit f73f5e7f (2018-05-22): x230: README: add plenty of hardware vendor links

Closes #24
closed 2018-05-22 by merge