merge/skulls #28: Possible to use internal flashing on non-heads release?

gazny, 2018-05-17:
I compiled coreboot for myself a while back. I'm looking into the project and would like to know if I can upgrade my coreboot to a skulls release. The documentation suggest I run the external scripts before I run prepare_internal_flashing.sh.

My question is can I up grade my bios by adding iomem=relaxed to my kernel parameters and run the script?

Also how would I be able to check if my bios is unlocked and eligible for the upgrade?

If this has been answered my apologies, looked at closed issues and I couldn't find information on this.

nemanjan00, 2018-05-17:
Skulls is just a little better updated CoreBoot with a lot simpler
instructions. If you have your Bios unlocked for internal flashing, then
yes. You can use that kernel parameter to allow fastboot to flash skulls.

merge, 2018-05-17:
It depends on whether you currently have the IFD locked. flashrom should simply fail if you have, so you can go ahead and try to run `prepare_internal_flashing.sh`. That script should probably simply be called `update.sh` or something.

gazny, 2018-05-18:
I ran dmidecode and got these results.
`Getting SMBIOS data from sysfs.
SMBIOS 2.7 present.
10 structures occupying 601 bytes.
Table at 0xBFF16020.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: coreboot
	Version: CBET4000 4.6-2479-g8e4384d0b4
	Release Date: 12/25/2017
	ROM Size: 4096 kB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 4.0
	Firmware Revision: 0.0

`

Followed by that I ran the prepare script and got this error.
`61: [: Illegal number: 40
00
24
79
43
84
-e The installed original BIOS is very old. please upgrade before installing coreboot.
Please search for your SODIMM RAM and verify that it uses 1,5 Volts (not 1,35V):
H641GU67F9333G
HMT451S6MFR8C-PB
`

Now I know the solution is to get my rpi and use my clips. but I am wondering if this effect is normal and If I can add any additional feedback for the project.

gazny, 2018-05-18:
Ah I see what my problem was. Thank you. The good news I ran the script and it worked effortlessly when I applied the proper syntax. Thank you for this great amount of work!