merge/skulls #30: Ability to set battery thresholds from coreboot config-tool payload

gazny, 2018-05-22:
I was wondering if we can set the battery thresholds similar to this [https://www.coreboot.org/Board:lenovo/x230#Setting_battery_threshold_on_a_X230_with_coreboot](url). I would like to be able to do this from the config payload if possible. But it may require additional dependencies if ectool is needed.

merge, 2018-05-23:
I think this would be a nvramcui development issue, but I doubt writing EC registers is in their scope either. I think the best fit would be to write a Linux driver for the x230's battery platform driver's sysfs interface.

Thrilleratplay, 2018-05-23:
@gazny  If you are a Linux user, I have heard that the v4.17 kernel will be have [this threshold patch](https://patchwork.kernel.org/patch/10205353/).  

Personally, this is a setting I would prefer to change in the user space for times those I want to have a full charge if I plan on using my battery for a period of time.

gazny, 2018-05-23:
I havnt seen many things on battery threshold intergation for corebooted x230s. The 4.17 kernnel hopfully will be sufficient.
This can be closed

closed 2018-05-23 by gazny