merge/skulls #32: add a second, more free release image: SeaVGABios with text-based graphics

merge, 2018-05-25:
What came out of #4 so far is that we could release an alternative config / image, without Intel's VGA Bios binary. For Linux users everything seems to work, but without framebuffer graphics, like we have it currently, only text-based. For GRUB users that means, `GRUB_TERMINAL=console` has to be in `/etc/default/grub`. A tested config already is the repo, named xxxxxxxxxx_free.config.

* Would you vote for releasing 2 images? I'm not sure about this.
* How would we minimize confusion and make the mentioned dependency very clear?
* I'm not sure about compatibility with Windows for the mentioned config.

gazny, 2018-05-25:
I vote yes. A step closer to a true freebios.

nemanjan00, 2018-05-25:
@gazny, I do not think we can get rid of gbe binary anyway...

Thrilleratplay, 2018-05-26:
Intel ME is the real obstacle to a libreboot BIOS for most machines, however GBE and the EC would still need to be included or rewritten for the x230.

@merge This is your project and I am probably out of line by saying this; providing a prebuilt image should available only as a backup.   From a security prospective, there is no difference between what you provide and sticking with Lenovo's stock BIOS.  I like the idea of this project because the scripts and configs help novices who would otherwise get stuck with outdated or incomplete tutorials.  If the emphasis is on building Coreboot and Heads with step by step instructions explaining what type of magic is happening behind the scenes, then the scripts could also give options to use the VGA binary or libgfxinit.  Building over prebuilt also allows other machines to be included, like the x220 that is far easier to build with ME/GBE/IFD included.

commit d1819537 (2018-11-23): x230: add a second image to the release package

We already have a config for a more free version using SeaVGABIOS. From now
on we build, test and release that too. The image filename should just
contail _free_ which is probably enough.

Closes #32
closed 2018-11-23 by merge