merge/skulls #37: Support changing SeaBIOS boot order

wlhlm, 2018-08-13:
I have my boot SSD (mSATA, Samsung) in one of the miniPCIe slots replacing the WWAN module and another drive (Crucial) in the 2.5" bay. SeaBIOS first tries to boot off of the drive in the bay, but that doesn't have any MBR and so it falls back to the payload (`nvramcui`). This means I explicitly have to invoke the boot menu using <kbd>Esc</kbd> to select the mSATA SSD.

Not sure if a way to change the boot order could be integrated into the `build.sh` script? At least I'd like to document the process here for someone with the same problem:

[SeaBIOS boot order documentation](https://www.coreboot.org/SeaBIOS#Configuring_boot_order)

SeaBIOS reads the boot order configuration from a "file" in [CBFS](https://www.coreboot.org/CBFS) which is part of the coreboot ROM. This file has to be called `bootorder` and includes a boot entry per line in a somewhat cryptic format (apparently inspired by Open Firmware) e.g. `/pci@i0cf8/usb@10,4/*@2`. This boot order file can be integrated into the coreboot build process using the `SEABIOS_BOOTORDER_FILE` Kconfig option.

The best way to get the boot order strings is to extract them from the coreboot debug log. This can be accessed using `cbmem` from coreboot-utils (output abbreviated):
```
# cbmem -c
[...]
Searching bootorder for: /rom@img/nvramcui
Searching bootorder for: /pci@i0cf8/*@1f,2/drive@0/disk@0
AHCI/0: Set transfer mode to UDMA-6
AHCI/0: registering: "AHCI/0: Crucial_CT750MX300SSD1 ATA-10 Hard-Disk (698 GiBytes)"
Searching bootorder for: /pci@i0cf8/*@1f,2/drive@2/disk@0
AHCI/2: Set transfer mode to UDMA-6
AHCI/2: registering: "AHCI/2: Samsung SSD 860 EVO mSATA 500GB ATA-11 Hard-Disk (465 GiBytes)"
[...]

Press ESC for boot menu.

Select boot device:

1. AHCI/0: Crucial_CT750MX300SSD1 ATA-10 Hard-Disk (698 GiBytes)
2. AHCI/2: Samsung SSD 860 EVO mSATA 500GB ATA-11 Hard-Disk (465 GiBytes)
3. Payload [nvramcui]

t. TPM Configuration
[...]
```
The important lines here start with "Searching bootorder":
```
# cbmem -c | grep "Searching bootorder"
Searching bootorder for: /pci@i0cf8/pci-bridge@1c/*@0
Searching bootorder for: /rom@img/nvramcui
Searching bootorder for: /pci@i0cf8/*@1f,2/drive@0/disk@0
Searching bootorder for: /pci@i0cf8/*@1f,2/drive@2/disk@0
Searching bootorder for: HALT
```
From the first `cbmem` output above I can induce that `/pci@i0cf8/*@1f,2/drive@0/disk@0` refers to the Crucial SSD and `/pci@i0cf8/*@1f,2/drive@2/disk@0` to my mSATA boot SSD from Samsung. To boot from the mSATA SSD by default, I have to create a `bootorder` file with something like this:
```
/pci@i0cf8/*@1f,2/drive@2/disk@0
/pci@i0cf8/*@1f,2/drive@0/disk@0
/rom@img/nvramcui
```
- - -
**Note:** I have yet to find the time to flash and try out this configuration. I will report back once I've done so.

merge, 2018-10-17:
I'd like to have this integrated into `x230_skulls.sh` as one simple option to switch to the mSATA being the default. This will involve:
* adding cbfstool to the `util` directory, as self-contained as possible.

If the mSATA option is chosen:
* build it as part of `x230_skulls.sh` (similar to ifdtool used in external_install_bottom.sh)
* generate the bootorder file you provided directly from the script
* let cbfstool add it to the `output/x230_coreboot_seabios_xxx_12M_prepared.rom` file we already generate

Feel free to do it and create a pull request. I might look into it too if time permits.

merge, 2019-04-08:
purism already does that, see https://source.puri.sm/coreboot/releases/blob/master/update_coreboot.sh

we could actually use quite some more from this script too.