merge/skulls #38: Skip RAM voltage check?

oida, 2018-09-25:
Hey, I was wondering what this is actually about: https://github.com/merge/skulls/blob/master/x230/x230_before_first_install.sh#L70
Since I was having a quite intense discussion with a friend about that, and if this may be needed at all.
From what we found in the wiki page which is linked: https://www.coreboot.org/Intel_Native_Raminit#Sandybridge.2FIvybridge I understood that the RAM voltage can only be adjusted on some Sandybridge/Ivybridge boards, but support for that is not implemented within coreboot, so RAM voltage is always fixed to 1,5V on the X230 and all Sandybridge/Ivybridge boards.
We also did a lookup and understood that PC3 is always 1,5V and all PC3L modules can run with either 1,35V or 1,50V https://en.wikipedia.org/wiki/DDR3_SDRAM#DDR3L_and_DDR3U_extensions so both PC3 and PC3L must work in the X230, which will always run with 1,5V.
Am I missing something or can this check be skipped alltogether, or at least for the X230? Thanks

merge, 2018-09-26:
All I know is that I've once plugged in a module specified with 1,35V and coreboot wouldn't start. So to me it made sense to add this in order to be safe. If you test this I'll do tests again and we'll see. But as long as it __could__ prevent failure it doesn't hurt to have this check in there I guess.

thanks for reviewing!

felixheld, 2018-11-16:
By specification DDR3L RAM has to work in 1.5V DDR3 slots; however DDR3 RAM might not work reliably at 1.35V. In general I wouln't trust information in the deprecated coreboot wiki too much.
If a SODIMM that works with the vendor firmware doesn't work with coreboot, it is due to some other bug. IIRC there was some problem with certain memory modules with the timC discovery; in that case trying to apply this patch might resolve the problem: https://review.coreboot.org/c/coreboot/+/23087

i-c-o-n, 2019-01-01:
I agree with Felix. If a DIMM is failing, it's not about the voltage (unless the DIMM is faulty somehow).

commit 717cf0e8 (2019-01-09): x230: skip ram voltage check

This simplifies setup by remove the unneeded RAM voltage checking.

According to the spec, DDR3L RAM has to work using 1.5V, see
https://en.wikipedia.org/wiki/DDR3_SDRAM#DDR3L_and_DDR3U_extensions.
Coreboot simply doesn't always implement changing voltage, which
can never prevent RAM from working though.

Closes #50
Fixes #38
closed 2019-01-09 by merge