merge/skulls #39: Enable USB always on?

MerlijnWajer, 2018-10-05:
For a while now, USB always on is supported in coreboot: https://review.coreboot.org/cgit/coreboot.git/commit/src/mainboard/lenovo/x230/cmos.default?id=7ffb329f278d6b027bb3b3660b69e87f1ddd69d8

But the default config does not enable it. Would it make sense to enable here? I personally find it a very useful feature.

merge, 2018-10-05:
sure. I'd be happy about a patch. Otherwise I might look into it next week. thanks

MerlijnWajer, 2018-10-05:
The patch would be only on the skulls build side, right? Not a coreboot patch? I am assuming it would be something like `sed -i 's/usb_always_on=Disable/usb_always_on=Enable/' src/mainboard/lenovo/x230/cmos.default` ?

merge, 2018-10-08:
@MerlijnWajer changing a git-tracked file in coreboot is not really an option. We could think about putting a patch up for review to enable this by default...

In the meantime: Can we configure this at build-time? Or do we have to build and ship nvramtool and let the user (a script) run it after flashing?

thanks

MerlijnWajer, 2018-10-09:
Putting up a patch for review that enables it by default would work. I think the build-time configuration simply requires changing the `cmos.default` file.

That said, I've been trying to make this work with nvramtool, and it doesn't work yet for me. Might be because I disabled ME (but this is maybe not the right place to figure out why it's not working):

```
# ./nvramtool -a | grep usb
usb_always_on = AC and battery
```

merge, 2018-10-10:
I can confirm that `./nvramtool -w usb_always_on="AC and battery"` (or "AC only") doesn't work -.- Unless there's a bug in nvramtool I doubt that changing the default file would help...

MerlijnWajer, 2018-10-10:
Sounds like that will first need to be resolved, then... Just to check - do you have the ME removed/disabled?

merge, 2018-10-10:
I have `me_cleaner -S` applied.

merge, 2018-11-13:
this might be related here. Maybe we can test that for the x230: https://ticket.coreboot.org/issues/171

MerlijnWajer, 2018-11-14:
Maybe it would be good to add this information to the ticket (that revert on the x230, doesn't help?). Thanks for testing, btw.

merge, 2019-01-18:
actually, it _is_ charging during standby for me...

MerlijnWajer, 2019-01-18:
Let me try the latest release and confirm...

closed 2019-01-20 by merge