merge/skulls #42: Random reset settings nvramcui

khseal, 2018-11-07:
Random resets settings after a reboot.

khseal, 2018-11-09:
hmm, thinkpad_acpi
This linux driver can have some bad interactions with the fallback/normal mecanism: when using it with the volume_control=1 option, volume_mode=1 is required, otherwise after shutting down the computer, it will always boot from fallback.

merge, 2018-11-13:
We don't use coreboot's [fallback mechanism](https://www.coreboot.org/Fallback_mechanism). If others would like to have that, it's up to them to create a PR and discuss it, but for now this is left to people who build coreboot themselves anyways.

Do you have a different problem here? If so, how can I reproduce it? thanks!

khseal, 2018-11-13:
I do not know what the problem is. The settings are simply reset randomly and it does not depend on which operating system I use.
I also had problems with the clock in Windows 10. The clock was transferred three hours ago. When I tried to solve this problem they were translated three hours ahead.

gazny, 2018-11-20:
Is this issue present for you in the newest release?

merge, 2019-01-20:
I'll clean up and close this as it doesn't seem to be a current issue. Please reopen or create a new issue in case there are still problems with nvramcui.

closed 2019-01-20 by merge