merge/skulls #43: Add support nvramtool

khseal, 2018-11-07:
nvramtool: CMOS option table not found in coreboot table.  Apparently, the coreboot installed on this system was built without specifying HAVE_OPTION_TABLE.

merge, 2018-11-13:
What's the issue here? I can run nvramtool without problems and HAVE_OPTION_TABLE is configured, see https://github.com/merge/skulls/blob/master/x230/config-715cb40963#L635

Do have `iomem=relaxed` on your linux cmdline? Since we read /dev/mem this might be necessary but I'm not sure. If not, how can I reproduce your problem?

thanks!

khseal, 2018-11-13:
I tried running the nvramtool program in lubuntu 18.04. I got this error.

merge, 2018-11-13:
and how did `cat /proc/cmdline` and `uname -r` look like?

khseal, 2018-11-13:
BOOT_IMAGE=/boot/vmlinuz-4.15.0-38-generic root=UUID=2b244152-70cc-46a3-8907-2ac4089131ae ro quiet splash vt.handoff=1

merge, 2018-11-14:
well, as I said. Try to add iomem=relaxed. nvramtool seems to work...

khseal, 2018-11-19:
Ok, thanks, problem solved

closed 2018-11-19 by khseal