merge/skulls #45: live USB system for flashing

merge, 2018-11-22:
What'd be nice to have is a live USB system with
* persistent (home) for including the Skulls files ( easy )
* have flashrom installed ( https://www.flashrom.org/Live_CD )
* boots Linux with iomem=relaxed boot option ( I guess we'd need to build one ourselves...)

If you have experience with creating live iso images, please help.

testbird, 2019-05-28:
The MX Linux live CD (debian stable) comes with a remaster tool.
You may just boot, adapt as needed and save the system to a new live cd image.
https://mxlinux.org/wiki/help-files/help-mx-remaster/

pahakalle, 2019-05-28:
@merge We could make a custom Alpine iso.
just need to add flashrom and you can also add iomem=relaxed to the kernel_cmdline part
https://wiki.alpinelinux.org/wiki/How_to_make_a_custom_ISO_image_with_mkimage

asvc, 2020-08-26:
Is this related to:

> create a live USB Linux for flashing including Skulls, flashrom and iomem=relaxed ( https://github.com/ivandavidov/minimal )

...and if "yes", is there a plan to work on it in 2020? Thank you.