merge/skulls #46: missing early linux console output when using SeaVGABIOS

merge, 2018-11-23:
When building using our "free" config (mv `config-xxx backup-config-xxx && mv free-config-xxx config-xxx && ./build.sh`) we don't have early linux boot console output on the display.

* How does GRUB behave when `GRUB_TERMINAL=console` is set?
* Can we find or file a bugreport in seabios for this?

Also, GRUB doesn't use the full screen in the framebuffer, but that's probably fine. It uses the same resolution as with Intel's vga bios binary.

paulmenzel, 2018-11-23:
On 11/23/18 11:29, Martin Kepplinger wrote:
> When building using our "free" config (mv `config-xxx
> backup-config-xxx && mv free-config-xxx config-xxx && ./build.sh`) we
> don't have early linux boot console output on the display.

Do you have the coreboot framebuffer driver (GOOGLE_FRAMEBUFFER_COREBOOT)
selected in the Linux configuration?

[…]

merge, 2018-11-23:
I didn't, but changed that now which doesn't help though. It feels like GRUB needs a change here because it's "loading init ramdisk" keeps displayed until Linux graphics takes over.

merge, 2018-11-26:
I'm running Debian stable's GRUB 2.02~beta3-5+deb9u1

dmesg is ok and unchanged. It's only about display behaviour during booting. With SeaVGABIOS GRUB is being displayed until Linux does modesetting and systemd's messages are displayed.

I only suspect a GRUB config might change that. I'm sure at all. When using Intel's vga bios, all early kernel messages are displayed in low resolution until Linux switches graphics mode.