merge/skulls #51: Can't resume from hibernate (suspend to disk)

varac, 2019-01-07:
I use skulls 0.1.0 and its released non-free images. Everything works so but *but* not resuming from hibernate. These are the regarding boot msgs trying to resume:

```
Jan 03 11:42:16 rocinante kernel: PM: Image signature found, resuming
Jan 03 11:42:16 rocinante kernel: PM: resume from hibernation
Jan 03 11:42:16 rocinante kernel: Freezing user space processes ... (elapsed 0.001 seconds) done.
Jan 03 11:42:16 rocinante kernel: OOM killer disabled.
Jan 03 11:42:16 rocinante kernel: PM: Marking nosave pages: [mem 0x00000000-0x00000fff]
Jan 03 11:42:16 rocinante kernel: PM: Marking nosave pages: [mem 0x0009f000-0x000fffff]
Jan 03 11:42:16 rocinante kernel: PM: Marking nosave pages: [mem 0xbff0d000-0xffffffff]
Jan 03 11:42:16 rocinante kernel: PM: Basic memory bitmaps created
Jan 03 11:42:16 rocinante kernel: PM: Using 3 thread(s) for decompression
Jan 03 11:42:16 rocinante kernel: PM: Loading and decompressing image data (702020 pages)...
Jan 03 11:42:16 rocinante kernel: Hibernate inconsistent memory map detected!
Jan 03 11:42:16 rocinante kernel: PM: Image mismatch: architecture specific data
Jan 03 11:42:16 rocinante kernel: PM: Read 2808080 kbytes in 0.01 seconds (280808.00 MB/s)
Jan 03 11:42:16 rocinante kernel: PM: Error -1 resuming
Jan 03 11:42:16 rocinante kernel: PM: Failed to load hibernation image, recovering.
Jan 03 11:42:16 rocinante kernel: PM: Basic memory bitmaps freed
Jan 03 11:42:16 rocinante kernel: OOM killer enabled.
Jan 03 11:42:16 rocinante kernel: Restarting tasks ... done.
Jan 03 11:42:16 rocinante kernel: video LNXVIDEO:00: Restoring backlight state
Jan 03 11:42:16 rocinante kernel: PM: resume from hibernation failed (-1)

```

My dmidecode details:

```
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 2.7 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: coreboot
	Version: CBET4000 4.8-2601-g2ca2acc51f-dirty
	Release Date: 12/19/2018
	ROM Size: 12288 kB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		BIOS is upgradeable
		Selectable boot is supported
		ACPI is supported
		Targeted content distribution is supported
	BIOS Revision: 4.0
	Firmware Revision: 0.0
```

There's also [S3 breaks system memory](#41) but that issue refers to sleep mode (suspend to ram).

merge, 2019-01-07:
I cannot reproduce this and I'm resuming fine from hibernation. I guess you have a different issue. What version of Linux are you running?

varac, 2019-01-07:
I'm using Ubuntu 18.04, kernel 4.18.0-13-generic.
It worked fine for be before flashing coreboot.

apraile, 2019-01-09:
> I use skulls 0.1.0 and its released non-free images. Everything works so but _but_ not resuming from hibernate. These are the regarding boot msgs trying to resume:
>
> ```
> Jan 03 11:42:16 rocinante [...]

It works for me if I run `systemctl hibernate` **with** root privileges, without root privileges I get the same boot messages and the resume fails.

Environment:

- Ubuntu 18.10
- 4.18.0-13-generic kernel
- systemd v239

varac, 2019-01-15:
> It works for me if I run systemctl hibernate with root privileges, without root privileges I get the same boot messages and the resume fails.

Same for me. I guess my initial issue was failing resume from hibernate, when I hibernated without sudo.
I'm now using the hibernate option from the gnome menu and everything is fine.

For me this issue can get closed. @apraile  ?

apraile, 2019-01-15:
> For me this issue can get closed. @apraile ?

Yes, also for me :ok_hand:

closed 2019-01-15 by varac