merge/skulls #55: Issues with top flash .sh

DanitiKain, 2019-01-15:
sudo ./external_install_top.sh -k x230xbackuptop.bin
No image file found. Please add -i <file>

Skulls for the X230
  Run this script on an external computer with a flasher
  connected to the X230's top chip (closer to the display
  and farther from you)

Usage: ./external_install_top.sh -i <image.rom> [-c <chipname>] [-k <backup_filename>]

 -f <hardware_flasher>   supported flashers: rpi, ch341a
 -i <image>              path to image to flash
 -c <chipname>           to use for flashrom
 -k <backup>             save the current image as
 -b <spi frequency>      frequency of the RPi SPI bus in Hz. default: 128

DanitiKain, 2019-01-15:
I got these instructions from here maybe its missing the -i option in the instructions.

DanitiKain, 2019-01-15:
I figured it out. I clone git repository and didn't find the bios files you were talking about until I went to where I extracted the tar file. So there I found the seabio files you were talking about. So I moved the scripts you fixed over and everything is working now. I think making the instructions just a slight bit more clear may help.

merge, 2019-01-15:
hi. thanks for reporting! are you sure you used the latest released version? the `-i <image>` option should really be optional and you should be able to choose one of our 2 released images; directly after extracting the tar archive.

DanitiKain, 2019-01-15:
yes I used the latest version snd grabed from git hub.

merge, 2019-01-15:
Nothing seems to have changed in this part of the script between now and the 0.1.0 release -.- but since copying today's script over to the 0.1.0 release directory fixes your problem, the upcoming release will fix your problem.

Can anybody reproduce this?

DanitiKain, 2019-01-16:
To be clear.  The problem was not fixed by just copying it over. I had to use the -i option even with the latest version that I cloned down to my machine.

merge, 2019-01-18:
what's the filename of your image? Did you change that? thanks.

DanitiKain, 2019-01-19:
I did not change it.  Was the same file name that I extracted from the tar file in 0.1.0 release. All I did was git clone the updates and then copy over the two .sh files to the untar directory for 0.1.0 release. When I get back to the laptop I was working on this on I will try and remember to update this post with the file name.

DanitiKain, 2019-01-20:
x230_coreboot_seabios_2ca2acc51f_top.rom
x230_coreboot_seabios_2ca2acc51f_top.rom.sha256
x230_coreboot_seabios_free_2ca2acc51f_top.rom
x230_coreboot_seabios_free_2ca2acc51f_top.rom.sha256

These are the file name.

merge, 2019-02-18:
we need someone to reproduce this. I recently used the script again and couldn't find the mistake.

merge, 2020-03-27:
please re-open in case you still have this problem, ok?

closed 2020-03-27 by merge