merge/skulls #57: Swap fn and ctrl key

DanitiKain, 2019-01-15:
I would love if one of the bios already had the fn and ctrl key swapped. Chatting in the coreboot irc channel and searching the web has not produce much results in how to make this happen. So just having it ready for those that one it from the get go would be great. I feel it is something that most thinkpad owners change more than any other bios setting. The other option I think would be creating a resource on this project for how to do it if you don't want to include it in the bios release would be very helpful for those that want to do it.

explit7, 2019-01-15:
keys settings are stored not in the BIOS, they are stored in the EC (Embedded controller)
You can patch Embedded Controller Settings (also Fn - Ctrl Key swap) as long, as you have Original Lenovo BIOS Installed.
After you patched it - the change will be stored permanently.
Afterwards you can install Heads, Skulls or Coreboot.

For patching see here:
https://github.com/hamishcoleman/thinkpad-ec

DanitiKain, 2019-01-15:
Ok before flashing skullls I had this setting saved. Should it be saved or do I now have to flash to original bios back and set the settings via link above. Also if I have to flash back what are the risks of doing this?

merge, 2019-01-15:
if you already have this EC modification _now_, this will be saved if you flash a coreboot BIOS like Skulls. We don't change the EC firmware here.

DanitiKain, 2019-01-16:
Ok to be more clear I was running stock bios. I had this settings set before flashing with Skulls. I have 4 thinkpads and I have all of them set this way. After flashing I booted into OS and it kept not working I thought it had messed the keyboard up then I realized ahh maybe I need to set in the bios. Now that is why I am here because it is so easy to just set. So if the settings from old Lenovo Bios EC settings should apply then that is an issue because they are not applying at the moment after I flashed skulls.

apraile, 2019-01-18:
Try this:

- In the SeaBIOS boot menu, select `2. Payload [nvramcui]`
- Enable `fn_ctrl_swap` option

The configuration is maintained between reboots. It works for me.

DanitiKain, 2019-01-19:
@apraile wow it was really that easy. How did #coreboot on IRC channel not know this.  Everyone was pointing to the above https://github.com/hamishcoleman/thinkpad-ec and talking about patching.

merge, 2019-01-19:
should we link to https://www.coreboot.org/Payloads#nvramcui which doesn't have much information either?

maybe try to create an official documentation page for nvramcui in coreboot? that might take time...

closed 2019-01-20 by merge