merge/skulls #59: x230: show bootsplash image when using SeaVGABIOS / native graphics init

merge, 2019-01-17:
As [found](https://github.com/merge/skulls/issues/21#issuecomment-453579034) in the docs, our nonfree coreboot config that uses Intel's VGA BIOS blob and configures coreboot to run the Option ROM, the bootsplash.jpg image is built-in.

That does not (yet) work for our free config, that uses native graphics init and SeaVGABIOS. For an overview, let's look at the current config-difference:

```
@@ -21,15 +21,14 @@
 CONFIG_INCLUDE_CONFIG_FILE=y
 CONFIG_COLLECT_TIMESTAMPS=y
 # CONFIG_TIMESTAMPS_ON_CONSOLE is not set
-CONFIG_USE_BLOBS=y
+# CONFIG_USE_BLOBS is not set
 # CONFIG_COVERAGE is not set
 # CONFIG_UBSAN is not set
 # CONFIG_NO_RELOCATABLE_RAMSTAGE is not set
 CONFIG_RELOCATABLE_RAMSTAGE=y
 CONFIG_CACHE_RELOCATED_RAMSTAGE_OUTSIDE_CBMEM=y
 # CONFIG_UPDATE_IMAGE is not set
-CONFIG_BOOTSPLASH_IMAGE=y
-CONFIG_BOOTSPLASH_FILE="bootsplash.jpg"
+# CONFIG_BOOTSPLASH_IMAGE is not set
 
 #
 # Mainboard
@@ -95,7 +94,7 @@
 CONFIG_VGA_BIOS_ID="8086,0166"
 # CONFIG_ONBOARD_VGA_IS_PRIMARY is not set
 CONFIG_DIMM_SPD_SIZE=256
-CONFIG_VGA_BIOS=y
+# CONFIG_VGA_BIOS is not set
 CONFIG_VGA_BIOS_FILE="pci8086,0166.rom"
 CONFIG_C_ENV_BOOTBLOCK_SIZE=0x10000
 CONFIG_MAINBOARD_PCI_SUBSYSTEM_VENDOR_ID=0x17aa
@@ -187,7 +186,6 @@
 # SoC
 #
 CONFIG_CPU_SPECIFIC_OPTIONS=y
-# CONFIG_S3_VGA_ROM_RUN is not set
 CONFIG_HEAP_SIZE=0x4000
 CONFIG_EHCI_BAR=0xfef00000
 CONFIG_SERIRQ_CONTINUOUS_MODE=y
@@ -294,6 +292,7 @@
 # CONFIG_NATIVE_RAMINIT_IGNORE_MAX_MEM_FUSES is not set
 # CONFIG_NATIVE_RAMINIT_IGNORE_XMP_MAX_DIMMS is not set
 CONFIG_SANDYBRIDGE_IVYBRIDGE_LVDS=y
+CONFIG_IF_NATIVE_VGA_INIT=y
 CONFIG_MAX_PIRQ_LINKS=4
 
 #
@@ -452,60 +451,21 @@
 # Devices
 #
 CONFIG_HAVE_VGA_TEXT_FRAMEBUFFER=y
-CONFIG_HAVE_VBE_LINEAR_FRAMEBUFFER=y
+CONFIG_HAVE_LINEAR_FRAMEBUFFER=y
 CONFIG_MAINBOARD_HAS_NATIVE_VGA_INIT=y
 # CONFIG_MAINBOARD_FORCE_NATIVE_VGA_INIT is not set
 CONFIG_MAINBOARD_HAS_LIBGFXINIT=y
-# CONFIG_MAINBOARD_DO_NATIVE_VGA_INIT is not set
+CONFIG_MAINBOARD_DO_NATIVE_VGA_INIT=y
 # CONFIG_MAINBOARD_USE_LIBGFXINIT is not set
-CONFIG_VGA_ROM_RUN=y
+# CONFIG_VGA_ROM_RUN is not set
 # CONFIG_NO_GFX_INIT is not set
-# CONFIG_ALWAYS_LOAD_OPROM is not set
-# CONFIG_ON_DEVICE_ROM_LOAD is not set
-# CONFIG_PCI_OPTION_ROM_RUN_REALMODE is not set
-CONFIG_PCI_OPTION_ROM_RUN_YABEL=y
-# CONFIG_YABEL_PCI_ACCESS_OTHER_DEVICES is not set
-CONFIG_YABEL_VIRTMEM_LOCATION=0x1000000
-# CONFIG_YABEL_DIRECTHW is not set
 # CONFIG_MULTIPLE_VGA_ADAPTERS is not set
 
 #
 # Display
 #
-CONFIG_FRAMEBUFFER_SET_VESA_MODE=y
-# CONFIG_FRAMEBUFFER_VESA_MODE_100 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_101 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_102 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_103 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_104 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_105 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_106 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_107 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_108 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_109 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_10A is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_10B is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_10C is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_10D is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_10E is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_10F is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_110 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_111 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_112 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_113 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_114 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_115 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_116 is not set
-CONFIG_FRAMEBUFFER_VESA_MODE_117=y
-# CONFIG_FRAMEBUFFER_VESA_MODE_118 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_119 is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_11A is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_11B is not set
-# CONFIG_FRAMEBUFFER_VESA_MODE_USER is not set
-CONFIG_FRAMEBUFFER_VESA_MODE=0x117
-CONFIG_BOOTSPLASH=y
 # CONFIG_VGA_TEXT_FRAMEBUFFER is not set
-CONFIG_VBE_LINEAR_FRAMEBUFFER=y
+CONFIG_GENERIC_LINEAR_FRAMEBUFFER=y
 CONFIG_LINEAR_FRAMEBUFFER=y
 # CONFIG_SMBUS_HAS_AUX_CHANNELS is not set
 CONFIG_PCI=y
@@ -585,7 +545,7 @@
 # CONFIG_MAINBOARD_HAS_I2C_TPM_GENERIC is not set
 # CONFIG_PLATFORM_USES_FSP2_0 is not set
 # CONFIG_INTEL_DDI is not set
-# CONFIG_INTEL_EDID is not set
+CONFIG_INTEL_EDID=y
 CONFIG_INTEL_INT15=y
 CONFIG_INTEL_GMA_ACPI=y
 # CONFIG_INTEL_GMA_SSC_ALTERNATE_REF is not set
@@ -713,8 +673,10 @@
 # CONFIG_SEABIOS_MASTER is not set
 # CONFIG_SEABIOS_REVISION is not set
 # CONFIG_SEABIOS_THREAD_OPTIONROMS is not set
+CONFIG_SEABIOS_VGA_COREBOOT=y
 CONFIG_SEABIOS_BOOTORDER_FILE=""
 # CONFIG_SEABIOS_ADD_SERCON_PORT_FILE is not set
+CONFIG_PAYLOAD_VGABIOS_FILE="payloads/external/SeaBIOS/seabios/out/vgabios.bin"
 CONFIG_SEABIOS_DEBUG_LEVEL=-1
 
 #
@@ -758,13 +720,11 @@
 # CONFIG_DEBUG_MALLOC is not set
 # CONFIG_DEBUG_ACPI is not set
 # CONFIG_DEBUG_CONSOLE_INIT is not set
-# CONFIG_X86EMU_DEBUG is not set
 # CONFIG_DEBUG_SPI_FLASH is not set
 # CONFIG_TRACE is not set
 # CONFIG_DEBUG_BOOT_STATE is not set
 # CONFIG_DEBUG_ADA_CODE is not set
 # CONFIG_HAVE_EM100_SUPPORT is not set
-CONFIG_NO_EDID_FILL_FB=y
 # CONFIG_ENABLE_APIC_EXT_ID is not set
 CONFIG_WARNINGS_ARE_ERRORS=y
 # CONFIG_POWER_BUTTON_DEFAULT_ENABLE is not set

```

I simply _quickly_ tried to accieve a similar configuration without blobs, but failed. I don't yet know at all why; I might very well just have used something wrong or have not spent enough time on this.

Do you have configured something similar already or know how/if this should work? thanks.

paulmenzel, 2019-01-17:
A verbose SeaBIOS log would be helpful. It can be acquired from CBMEM console with the utility `cbmem -c`.

jellolag, 2020-03-28:
Could one run coreboot in an emulator and then capture via hooks, the configuration the Intel VGA bios performs in order to map it to the SeaVGA bios?

I haven't read the coreboot source yet. Just uboot for my  toshiba ac100 to enable a boot menu and larger resolution screen.

m4rc0linux, 2020-11-13:
I was playing with the scripts and I made it work rebuilding skulls as per the instructions with a simple modification on the free-defconfig-8053595370.
The problem seems to be related to the libgfxint supported resolution. When not set (like in the default config) it will try get the maximum which is 1366x768 (for the x230) and the image will not show or shown incorrectly. I have tried all possible combinations (to leave like default and to set the resolution to 1366x768 and to use a 1366x768 jpg and the default 1024x768) but with no good results. I think the max supported is 1024x768 - also the default skulls bootsplash is an 1024x768 jpg.
Nothing more than adding in to the free-defconfig-8053595370 something like:
CONFIG_LINEAR_FRAMEBUFFER_MAX_WIDTH=1024
CONFIG_LINEAR_FRAMEBUFFER_MAX_HEIGHT=768
then, I have added the bootsplash image as per the coreboot instructions https://www.coreboot.org/SeaBIOS#Adding_a_graphical_.22bootsplash.22_image    
However I think that adding on the script something like 
CONFIG_BOOTSPLASH_IMAGE=y
will make the image to be built into the rom (but I didn't try it yet).
Using this workaround the bootsplash is shrinked and displays to a lower resolution, however it works, and it is fine for me.. I am finally out of nonfree/vga version...
Thank you guys for this fantastic project!

rnhmjoj, 2020-12-22:
@m4rc0linux I tried following your directions. I'm using libgfxinit and I confirm that 1024x768 is the highest resolution supported. If I try higher resolutions the jpeg decoder fails with [ERR_WIDTH_MISMATCH](https://github.com/coreboot/coreboot/blob/7128063ecc8099932a4bef8d2e28101b7f652804/src/lib/jpeg.h).
However, in my case, the image is not stretched horizontally and the color channels are [swapped](https://ticket.coreboot.org/issues/287). The color issue can be worked around, but not the stretching.

In the end, I think I'll take the full resolution with no splash over this.

closed 2021-03-21 by merge