merge/skulls #62: test building and running yabits instead of SeaBIOS

merge, 2019-01-29:
https://yabits.github.io/ seems to be a cool Project providing a UEFI coreboot payload. Why not build that and add a 3rd image to our x230 release? Any test results are welcome.

pgera, 2019-03-09:
With the recent switch to Mr. Chromebox's branch for tianocore, that may also be a viable alternative.

nroetert, 2020-07-26:
I would love to test this, complete noob though.
My X230 is currently on Skulls 0.1.9 with SeaBIOS, how do I build one with Yabits instead?

merge, 2020-07-26:
this is merely an idea one could implement.

Am 26. Juli 2020 17:26:44 MESZ schrieb Niels Roetert <notifications@github.com>:
>I would love to test this, complete noob though.
>My X230 is currently on Skulls 0.1.9 with SeaBIOS, how do I build one
>with Yabits instead?

-- 
Martin Kepplinger
xmpp: martink@jabber.at
sent from mobile