merge/skulls #65: GRUB framebuffer not using whole screen with libgfxinit

paulmenzel, 2019-02-03:


paulmenzel, 2019-02-03:
As document in the [how-to from Chuck Nemeth](https://www.chucknemeth.com/flash-lenovo-x230-coreboot/) changing the GRUB (in MBR) configuration like below, should do the trick.

```
[user@mainpc:~]$ sudoedit /etc/default/grub

# Uncomment and edit the following line to read:
GRUB_GFXMODE=1366x768x32

[user@mainpc:~]$ sudo update-grub
```

merge, 2019-02-04:
thanks for looking that up paul, that indeed works. we'll have to document this.

so grub's `vbeinfo` knows about this mode. why doesn't it use it by default?

paulmenzel, 2019-02-04:
On 04.02.19 22:04, Martin Kepplinger wrote:
> thanks for looking that up paul, that indeed works. we'll have to
> document this.

Thank you for testing this so quickly.

> so grub's `vbeinfo` knows about this mode. why doesn't it use it by
> default
Good question. Unfortunately, I do not know. Do you have the full output?

What GRUB version is in your MBR?

merge, 2019-02-04:
I didn't save the full output. Would that help? GRUB sees 640x480 which was used by default (in 2 color depth modes I think), and the mentioned correct screen resolution.

I use debian stable's version 2.02~beta3-5+deb9u1

paulmenzel, 2019-02-04:
On 04.02.19 22:37, Martin Kepplinger wrote:

[…]

> I use debian stable's version 2.02~beta3-5+deb9u1

It’d be great, if you tested this with the latest version from the 
master branch (or some Fedora or Ubuntu live system for example, as they 
have several GRUB patches applied too).

pgera, 2019-09-16:
It's not just grub; this also happens if you add tint as a payload. Also, memtest (the coreboot payload) doesn't display anything with libgfxinit for me. (I'm not using skulls, but I doubt if there is any difference)