merge/skulls #71: add a script to flash the vendor backup

merge, 2019-02-08:
Should we add a script to flash the vendor backup? If so:

* Does it make sense to create one for internal flashing?
* (Of course it would take the 2 backups as an input)

Baitinq, 2019-09-07:
I think that would be nice (if possible). Once the vendor backup is flashed would the write perms on the bios chip be removed or would they stay so it was possible to install coreboot again from software? (without manual flashing)

merge, 2019-09-07:
the flash would be locked again.

Baitinq, 2019-09-07:
What if there was an option in the script to choose either to keep the write perms or lock them down?

nroetert, 2020-07-26:
Did this actually happen, got added?

merge, 2020-07-26:
this is an idea one could do. not yet done.

Am 26. Juli 2020 17:24:10 MESZ schrieb Niels Roetert <notifications@github.com>:
>Did this actually happen, got added?

-- 
Martin Kepplinger
xmpp: martink@jabber.at
sent from mobile