merge/skulls #74: Consider adding Memtestx86 downloadable ROM

dendrobedan, 2019-02-17:
While it is true, that almost every Linux Distribution comes pre-equiped with Memtestx86 this would be a nice addition to be able to check for RAM-Issues without the need to boot from a Harddrive.
That is of course if space limitations aren't a Problem.

Thrilleratplay, 2019-02-17:
I can confirm that all secondary payloads can be added without space being an issue.

```
CONFIG_COREINFO_SECONDARY_PAYLOAD=y
CONFIG_MEMTEST_SECONDARY_PAYLOAD=y
CONFIG_NVRAMCUI_SECONDARY_PAYLOAD=y
CONFIG_TINT_SECONDARY_PAYLOAD=y
```

While [Tint](https://www.coreboot.org/Tint) may not seem necessary, there is no reason not to add it.

merge, 2019-02-19:
memtestx86 could be nice to have indeed. Let's keep this issue open and see if it gets any more attention ok?

Thrilleratplay, 2019-02-22:
@merge Given how trivial adding the remaining secondary payloads and the space is being unused otherwise, why not just add MemtestX86 and Tint?  They are there is the user want to use them and can be ignored otherwise.

PR #87 (2019-05-06): x230: update config / add memtest and coreinfo payloads
This patch adds two new payloads (memtestx86 and coreinfo)
to resolve #74.
commit 71130df3 (2019-05-06): x230: update config / add memtest and coreinfo payloads

This patch adds two new payloads (memtestx86 and coreinfo)
to resolve #74.
closed 2019-05-06 by merge