merge/skulls #77: Question about BIOS version

Smeat, 2019-02-23:
Hi I just tried updating my EC.
You write "In case you're not running the latest BIOS version, either[..]", which I interpreted as "do exactly one of those options".
So I chose the second option and applied the battery patch.
After updating, my EC is now Version 1.14, but my BIOS is still 1.13. Everything seems to be working fine, but `x230_before_first_install.sh` still complains, since it only checks for the BIOS version.

As I understand it I don't really need the newest BIOS, just the newest EC?

merge, 2019-02-23:
Hi, thanks for your report! that's strange. My understanding was that with thinkpad-ec you would update the bios too. when did you do that. They seem to have the latest bios version 2.74: https://github.com/hamishcoleman/thinkpad-ec/commit/80a5dd882a894709df675b4ff961e4bbe8885362 and that's what we check.

That said, we could easily relax our checking in that script a bit. We probably sould only check for 2.7x or something because the EC firmware never changes.

Smeat, 2019-02-23:
I did it today, so I was using the current master.
Maybe I am just a special case though, since I did use a rather unusual setup. Unfortunately my device has a supervisor password with USB boot disabled (but BIOS flashing is enabled. This might be the cause though), so I had to chainload to the USB device, on which I had to use syslinux with memdisk. But it did flash the EC successfully and I can't remember any error message. It did even show the BIOS and EC version (2.74 and 1.14).

Looking at the readme they suggest updating with the official version first: "For best results, ensure you have updated your BIOS to a recent version before starting". Maybe a similiar message should be in your documentation as well.

Do you know, if the BIOS version itself even matters when flashing coreboot? I'd say no, but I am not quite sure if I understand the whole architecture.

merge, 2019-02-23:
> I did it today, so I was using the current master.
> Maybe I am just a special case though, since I did use a rather unusual setup. Unfortunately my device has a supervisor password with USB boot [...]

or, easier, don't check for that very latest bios at all.

> 
> Do you know, if the BIOS version itself even matters when flashing coreboot? I'd say no, but I am not quite sure if I understand the whole architecture.

no it doesn't. and we should only check for the 1.14 EC version actually. Everything else should only be suggestions in the docs.

commit 8814c692 (2019-02-27): x230: remove checking for the very latest Lenovo BIOS

This makes things easier to maintain and simply causes less confusion
for users. We only want the lastest EC firmware.

Closes #77
closed 2019-03-03 by merge