merge/skulls #78: Boot speed

yegortimoshenko, 2019-02-26:
Skulls boot (both free and non-free VGA) seems to take ~2s longer than other Coreboot+SeaBIOS derivatives I have access to (APU2, Librem 13, Toshiba `swanky` Chromebook). What could be the reason behind that?

Judging by https://www.seabios.org/Runtime_config and https://github.com/coreboot/seabios/blob/14221cd86eadba82255fdc55ed174d401c7a0a04/src/boot.c#L466, there is `boot-menu-wait` option that is set to 2.5 seconds by default. Since it is always possible to hold down Esc prior to boot, maybe that delay could be set to 0?

merge, 2019-03-04:
I'm not sure. I think we really should leave users time to "press ESC". But there might still be ways to speed up things before that. Anyways, I'd be happy for hints or a patch :)

pgera, 2019-09-21:
On my t530, I see a difference in warm reboot (fast) v/s cold reboot(slow). I'm talking about the delay before the splash, not the "press Esc" delay. This may be due to waiting for keyboard init.

closed 2024-12-15 by lukateras