merge/skulls #81: add a reference to new ACPI methods to set battery thresholds in changelog

apraile, 2019-03-08:
With the new **skulls 0.1.2**  :tada: release and thanks to [this](https://review.coreboot.org/c/coreboot/+/23178) patch, it's possible to set battery thresholds easily in GNU/Linux systems without [ectool hack](https://www.coreboot.org/Board:lenovo/x230#Setting_battery_threshold_on_a_X230_with_coreboot). A minimum kernel version of 4.17 is required.
It would be nice to report it in the changelog.

merge, 2019-03-11:
that's awesome and while it's an additional feature compared to the vendor bios, it is a generic interface and ideally should be picked up by Desktop environments (if it isn't already) or other UIs to be visible to users, independently of hardware, shouldn't it?

At least I'm not sure where or whether we should document this. After all there are other coreboot related Linux features IIRC.

apraile, 2019-03-11:
Yes, on second thought, it would be more appropriate to document it in the official coreboot doc.
Thanks.
(I close the issue)

closed 2019-03-11 by apraile