merge/skulls #82: Chip name issue

mrxanto, 2019-04-03:
Hello.

When I try to execute a "external_install_top.sh" script I have error:

Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Quit
Please select the hardware flasher: 2
trying to detect the chip...
chip not detected.
flashrom v0.9.9-r1954 on Linux 4.15.0-47-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Eon flash chip "EN25QH32" (4096 kB, SPI) on ch341a_spi.
No operations were specified.
Please find it manually in the list above and rerun with the -c parameter.

As I understand, this error produced by string:
`CHIPNAME=$(cat` "${TEMP_DIR}"/chips | grep Found | grep "MX25L6406E/MX25L6408E" | grep -oP '"\K[^"\047]+(?=["\04
7])' || true)

Why is needed to grep only such chip model? As you see, on my board are installed "cFeon QH32 104HIP" chip.

merge, 2019-04-03:
We add the models because users otherwise would usually manually choose and run flashrom with "-c chipname". You can use our script that way now too (./externel_install_top.sh -c EN25QH32) Anyways, thanks for reporting this. This helps.

6r1zzlyB, 2019-04-03:
I has a similar issue.

```
pi@raspberrypi:~/skulls-x230-0.1.3` $ sudo ./external_install_bottom.sh -m -k oem__bios_2019-04-03_bottom.bak
Skulls for the X230

Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Exit
Please select the hardware flasher: 1
Ok. Run this on a Rasperry Pi.
trying to detect the chip...
Detected "EN25QH64".
make: Entering directory '/home/pi/skulls-x230-0.1.3/util/ifdtool'
make: Leaving directory '/home/pi/skulls-x230-0.1.3/util/ifdtool'
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom v1.1-rc1-1-gcb97368 on Linux 4.14.98-v7+ (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Error: Unknown chip '"EN25QH64"' specified.
Run flashrom -L to view the hardware supported in this flashrom
```

**Had to specify the chip in the command.**

```
pi@raspberrypi:~/skulls-x230-0.1.3 $ sudo ./external_install_bottom.sh -m -k oem__bios_2019-04-03_bottom.bak -c EN25QH64
Skulls for the X230

Please select the hardware you use:
1) Raspberry Pi
2) CH341A
3) Exit
Please select the hardware flasher: 1
Ok. Run this on a Rasperry Pi.
make: Entering directory '/home/pi/skulls-x230-0.1.3/util/ifdtool'
make: Leaving directory '/home/pi/skulls-x230-0.1.3/util/ifdtool'
Intel ME will be cleaned.
The flash ROM will be unlocked.
Start reading 2 times. Please be patient...
flashrom v1.1-rc1-1-gcb97368 on Linux 4.14.98-v7+ (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Eon flash chip "EN25QH64" (8192 kB, SPI) on linux_spi.
Reading flash... done.
flashrom v1.1-rc1-1-gcb97368 on Linux 4.14.98-v7+ (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Eon flash chip "EN25QH64" (8192 kB, SPI) on linux_spi.
Reading flash... done.
current image saved as oem__bios_2019-04-03_bottom.bak
connection ok
start unlocking ...

```

merge, 2019-04-03:
ah I see. that's with new new versions of flashrom I guess... but we'll look into it and I'm happy for help :) could you reopen the issue?

Am 3. April 2019 19:08:07 MESZ schrieb "Matthew St. Denis" <notifications@github.com>:
> I has a similar issue.
>
> ```
> pi@raspberrypi:~/skulls-x230-0.1.3` $ sudo ./external_install_bottom.sh
> -m -k oem__bios_2019-04-03_bottom.bak
> Skulls for the X230
>
> Please select the hardware you use:
> 1) [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

merge, 2019-04-03:
This was a real bug I guess. If you can, please test the current git master branch's version of the script. thanks again for reporting

commit 131baa6e (2019-04-03): x230: top: support detection for EN25QH32

This feels like the last piece of the puzzle in chip-detection.

Fixes #82
commit 7f3276b7 (2019-04-03): x230: external_install_bottom: fix detecting EN25QH64 chip

This should fix #82
closed 2019-04-03 by merge