merge/skulls #83: FHD mod question

6r1zzlyB, 2019-04-05:
Is there a way to build out the Skulls image with the FHD mod?

https://review.coreboot.org/c/coreboot/+/28950

This is my first time modding bios. I was able to flash Skulls without any issues. But I am still learning.

merge, 2019-04-08:
@6r1zzlyB  Do you have a full HD display installed? Looks like we could add support for that as a new Laptop basically. Just like for any new laptop support, we'd need somebody who has the hardware and time to keep documenting, building, maintaining and testing.

jyvet, 2019-05-02:
I'll do the nitrocaster's FHD-mod soon. I can then take care of the 'new laptop support'. What's the procedure to include a coreboot patch which is not merged upstream? I can slightly modify the bash script or... try to help to get it merged if this is necessary for skulls.

merge, 2019-05-03:
sounds good. For out-of-tree changes, most importantly put them up to review.cb.org.

Also, note that we want to have only one coreboot for all our configs/laptops! so in the end you want all you need merged.

It may not be trivial to include multiple changes, and they should definitely all be temporary, "until merged"! we'd find a way, look at the common download script....

Am 2. Mai 2019 22:49:32 MESZ schrieb Jean-Yves VET <notifications@github.com>:
>I'll do the nitrocaster's FHD-mod soon. I can then take care of the
>'new laptop support'. What's the procedure to include a coreboot patch
>which is not merged upstream? I can slightly modify the bash script
>or... try to help to get it merged if this is necessary for skulls. 

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

nikitaturkmen, 2019-09-17:
@merge can i use FHDmod with skulls.
http://wiki.muc.ccc.de/x230modding:start

merge, 2019-09-17:
I don't have the hardware, sorry. But you'll use the displayport interface, and that is of course usable with coreboot. I'm not sure how convenient it is to use.

Biont, 2020-01-01:
I have a FHD (nitrocaster) x230 and flashed coreboot via skulls on it. So I would be more than happy to help testing.

What needs to be done to move forward here?

Could you please throw  me a few pointers on how to extend the build scripts to build coreboot with the relevant patch(es)? I have no experience with coreboot itself and from looking at the build scripts, it seems like you're basically passing volumes and environment variables to a preconfigured docker container, so I am not sure at what point you could introduce source patches.

merge, 2020-01-02:
in general the way to apply a custom patch to coreboot would be to put it up for review to coreboot upstream and add it similar to how we did it in the past: https://github.com/merge/skulls/commit/1fdaf3797ba12580fd4e7491e4502e89ce0b4101

Biont, 2020-01-02:
Oh I see. Thank you. Shouldn't it then be enough to run the build script and point it to [this commit with the latest patchset](https://review.coreboot.org/plugins/gitiles/coreboot/+/48baddc90ca895443f41f411dc0433ab1aa1e2db)?

`./build.sh -c 48baddc90ca895443f41f411dc0433ab1aa1e2db x230` essentially?

Biont, 2020-01-03:
Okay here is what I did:

- I added a `-p | --patchset` command line argument that exports the `COREBOOT_PATCHSET` environment variable
- Added a `function checkoutPatchset()` to `download_coreboot.sh` that checks out coreboot at the specified patchset in `$COREBOOT_PATCHSET` 
- I copied the `x230` folder to `x230_coreboot` and modified its `build.sh` so that it checks out https://review.coreboot.org/c/coreboot/+/28950 with the new argument (along with renaming the defconfig files so that they no longer contain commit hashes)

This results in a successful build, but I still have to build the courage to actually flash it.

A few questions:
What do you think of this approach? Would you be interested in a PR with at least the script changes providing the ability to check out patchsets?

How would you structure the nitrocaster support in this project? Having it as a separate top-level directory made sense to me because I did not want to make large changes to the existing x230 setup (->defconfig filenames containing the git hashes needed to go).
But now it shares 99% of the folder contents for no real reason, so a different approach would make sense as well.

There was the following notice at the end of the build:

```

Built lenovo/x230 (ThinkPad X230)

        ** WARNING **
coreboot has been built without an Intel Firmware Descriptor.
Never write a complete coreboot.rom without an IFD to your
board's flash chip! You can use flashrom's IFD or layout
parameters to flash only to the BIOS region.

4+0 records in
4+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.00512383 s, 819 MB/s
==================== result: ======================
/home/coreboot/cb_build/coreboot_lenovo-x230-top.rom
===================================================

```

Should I be worried? Or is this all looking good to you @merge ?

Biont, 2020-01-25:
> A few questions:
What do you think of this approach? Would you be interested in a PR with at least the script changes providing the ability to check out patchsets?
How would you structure the nitrocaster support in this project? 

@merge Any ideas/opinions?

merge, 2020-01-30:
@Biont that all doesn't sound bad. I'd be happy to look at a MR with the mentioned addition for checking out a patch. If it's an optional addition and only the FHD laptop uses it (the same script then), fine.

I'd create a top level directory, yes. Just start by putting in only the config (and a few instructions in the commit messages) and we can work from there incrementally until we can package a release without too much duplication in the repo.

jyvet, 2020-01-30:
I've done the nitrocaster mod 2 days ago. Sorry for the long delay. I wanted to use a 13.3" FHD screen but had issues to adjust the screen bezel (I eventually broke the screen). Now I'm using a 12.3" FHD screen and also managed to apply the patch with skulls. Maybe it would be better to try to cherry-pick the patch instead of checking out the commit. If the patch applies we could automatically get newer coreboot versions working without additonal work. @Biont Tell me if you need some help to prepare the patch and/or test it.

nikitatikin, 2020-03-01:
hi any updates ??. I have hardware i can help you @Biont . Normally im using nitrocaster patched coreboot, but i cant boot my pc from usb i get error.

Biont, 2020-03-02:
Hi @nikitatikin, my branch and initial commit is [here](https://github.com/Biont/skulls/commit/eda783ab460dd4cf0f77be2c04abcdb6a884c467), but other than pushing my stuff into that fork, I have not yet found the time to to additional cleanup to prepare a PR, sorry.

If anyone willing and daring wants to try and help out, be my guest.

nikitatikin, 2020-03-12:
@Biont i cant build it. could you build and push please ? i get some file missing errors.

naggie, 2020-07-14:
I have a nitrocaster modded x230 -- happy to try as soon as there's something solid. Could this end up as part of skulls or always require a custom build process?

merge, 2020-07-14:
it could easily be part of skulls - if somebody does it. we've recently added "x230t" - just similarly add "x230_fhd" if you have the time to do it.

naggie, 2020-07-15:
Great -- I will certainly give it a go next time I get the x230 out, but don't have much time at the moment.

gulafaran, 2020-08-16:
any updates on this? im about to FHD mod my x230

naggie, 2020-08-28:
wow, @jyvet thanks for working on this. Even though I ended up just making a coreboot image for my x230-fhd from scratch, your implementation and readme is great interest to me. I've managed to get the splash screen somewhat working, but the image has the wrong colours and is surrounded by noise. It is not scaled, and has to be a VESA resolution image. I see you've added quite a bit about the bootsplash so hopefully it will help me with mine.

I'm writing a blog post about the process too.

jyvet, 2020-08-31:
I've just slightly modified the scripts to be able to automatically cherry pick the FHD patch. The Readme file was duplicated from the X230 board. I added a tiny section and removed some other ones. The work on bootsplash/Readme was mainly made by @merge then. The bootsplash should work out of the box. I'll start testing this week and will confirm that. In the meantime, let me know if you have some ideas on improving the patch! In the future I plan to get another X230 with nb51's mod to be able to test that as well. The patch should work with nb51's mod except the back-light control (which requires a script to be run from your OS to re-redirect to commands to a USB interface). I guess I'll prepare such script (for Linux first) and add it to Skulls.

gulafaran, 2020-08-31:
@jyvet thanks! i just recieved the board myself, so shall see when i attempt the soldering. was much smaller then i expected so a bit scared. but i will report back when i have tried your build scripts.

jyvet, 2020-08-31:
Good luck with the soldering! It should be fine. I used a magnifying glass and it makes it much easier. Though, mine is not perfect, the screen is flickering from time to time. I need to check if this is a cable issue or if I should use my soldering iron again.  
Maybe I should have put more soldering flux (I guess this is the most important step)! Anyway the result looks great. I hope one day we'll be able to install an OLED screen!

Thrilleratplay, 2020-08-31:
@gulafaran I posted [a few nitrocaster board installation tips/suggestions on Reddit a while back](https://www.reddit.com/r/thinkpad/comments/8hvdb1/fuck_this_is_much_smaller_than_i_thought/dyn0y4t/).  Please learn from my expensive mistakes.

jyvet, 2020-08-31:
I've just tested the binaries. There is a tiny issue I'm not sure to be able to fix: The splashscreen is displayed but when pressing ESC, the screen goes black and I'm not able to see the menu with the different payloads. Everything else seems to work. Any idea how to tweak coreboot to be able to display the boot menu?

naggie, 2020-08-31:
@jyvet how does the splash display? Is it stretched across the full screen, or windowed with noise around? (I get the latter)

I assume we're limited to VESA resolutions.

jyvet, 2020-09-01:
It is stretched across the full screen. No noise around. Are you able to display the list of payloads?

naggie, 2020-09-03:
Encouraging -- I don't see a list of payloads but I have only a single, SeaBIOS which it boots into no problem. I think I'm simply missing a config option or something -- I will diff mine against your skulls

fincham, 2021-02-08:
I switched from vanilla coreboot on my Nitrocaster rev6 X230 to @jyvet's `x230_fhd` fork (commit 
8cc80fa) with `x230_fhd_coreboot_seabios_f5b3a49b76_top.rom` and everything is working properly as far as I can tell. I had to convert from an EFI style partition layout (as I had been using GRUB payload before) to an MBR one and install `grub-legacy` but otherwise it all worked.

Tested:

 * Splash screen displays properly though it's a little fuzzy (I guess it's being scaled, I'd probably prefer no scaling)
 * Pressing <kbd>Esc</kbd> gives a list of other payloads and they all seem to work
 * Can boot Ubuntu 20.04 from a live USB, or Debian Buster from hard drive
 * `grub-legacy` from HDD comes up in 80x25 mode
 * Brightness adjustment works inside X
 * Suspending the laptop powers off the LCD

fincham, 2021-02-08:
One quirk I noticed, though it seems to be harmless:

I'm using an LG LP125WF4-SPB1 panel, and with a 5.9 kernel during boot I see:

```
[    2.566781] i915 0000:00:02.0: [drm] Panel advertises DPCD backlight support, but VBT disagrees. If your backlight controls don't work try booting with i915.enable_dpcd_backlight=1. If your machine needs this, please file a _new_ bug report on drm/i915, see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
```

The backlight control still works regardless.

jyvet, 2021-02-08:
Thanks for having tested the patch!

TheChuckster, 2023-05-08:
Any updates? Has this patch been included yet? As someone with the FHD mod myself, if it can help move things along, I would love to make a 3 or 4 digit $ USD donation to the project, thanks.

tlaurion, 2023-05-08:
This patch works against 4.19
https://github.com/osresearch/heads/blob/master/patches/coreboot-4.19/0001-x230-fhd-variant.patch

See associated coreboot config under heads and commit linked to it being merged for details.

As said under related PR, work needs to be pushed upstream under coreboot for that to be merged as a x230 variant.