merge/skulls #88: flash_bottom + bus pirate never ending?

Unclezz, 2019-05-11:
Hi everyone.
I slightly changed the external_install_bottom.sh in order to use a bus pirate instead of rpi.
Actually I simpy change the reference to the proper flashrom dev with "buspirate_spi:dev=/dev/ttyUSB0".

The reading of the bottom chip went through without issues and I have the backup too. However it is now stuck on:
`Erasing and writing flash chip...`

it's there for already 10 hours or more. I see USB led activity on the bus pirate but looks to me a bit too much time. 
how much should I expect to wait for the writing action? Is there anything you can suggest me to do in such situation?

Thanks

merge, 2019-05-13:
sorry to hear that. I'd not use the script, but directly run flashrom on the commandline in your case. Since you have the backup, just try to write an image again, using whatever flashrom spispeed and other settings that apply to your buspirate connection. You don't have to wait 10 hours. 15min should be enough I think.

Unclezz, 2019-06-08:
Hi,
I tried using manual commands with flashrom and also I updated the firmware of the Bus Pirate to the latest available. Unfortunately Bus Pirate is able to read the content of the flash but is not able to erase and write. It is also dramatically slow even just in reading phase.

I moved to CH341A and the flash was smooth and fast.

Thanks @merge for this great piece of work, now my X230 is running coreboot :):)

merge, 2019-06-10:
ok. good to hear.

closed 2019-06-22 by merge