merge/skulls #89: Immediate restart after loading rootfs

asvc, 2019-06-02:
"One gif worth thousand of words"
[image]
_slow-motion recording on mobile phone_

This currently prevents from using https://github.com/Drive-Trust-Alliance/sedutil/ with Skulls.
Any hints on how to debug it, perhaps re-compiling sedutil/PBA with the newer Syslinux will help?

Latest "Tails" boots-up fine, while for example "Rescatux" hangs at the beginning: both are based on Debian, 4.19.XX kernel.

asvc, 2019-06-05:
A bit of an update. I've re-flashed Skulls with proprietary Intel video blob and now able to read the following before computer restarts:

```
SYSLINUX 6.03 EDD 2014-10-06
Loading bzimage...ok
Loading rootfs.cpio.xz...ok
early console in extract_kernel
input data: 0x01660100
input_len: 0x005c7229
output: 0x01000000
output_len: 0x00b5df18
kernel_total_size: 0x00c42000

Decompressing Linux... Parsing ELF... No relocation needed... done.
Booting the kernel.
```
Values seems to be at the very least not completely random.
**input_len**: 0x005c7229 --> 757Kb
**kernel_total_size**: 0x00c42000 --> 1.6Mb

I don't think that option of entering "menu module" is enabled in the sedutil's Syslinux config, so I can't press <kbd>Tab</kbd> and modify options on the fly.

For the record, booting from the same flashdrive on another system running vendor's BIOS, echoes this output after _"Booting the kernel"_ line:
```
Starting logging: OK
Initializing random number generator... done
Starting network: OK

*************
* DTA sedutil rescue image...
```

asvc, 2019-06-08:
...small progress, [PBA](https://github.com/Drive-Trust-Alliance/exec/blob/master/biospba.img.gz) written directly to the USB drive boots successfully.  
Seems like the problem lies within interoperability of the particular  [Syslinux config](https://github.com/Drive-Trust-Alliance/sedutil/blob/master/images/buildroot/syslinux.cfg)  and Skulls.
 
Fun ensues.

asvc, 2019-06-08:
After modifying syslinux config to allow greater verbosity, I am observing **[1]** the following errors during the boot

```
mce: [Hardware Error]: CPU 0: Mahine check: 0 bank 5: ae000000004110a
mce: [Hardware Error]: TSC 0 ADDR c1aaae80 MISC b1e0000086
mce: [Hardware Error]: PROCESSOR 0:306a9 TIME 1787430704 SOCKET 0 APIC 0 microcode 21
```
This is x230i. @merge, is there anything i5-specific in Skulls that might prevent i3-based machines from operating normally?

**[1]** Using 240fps slow-motion recording.

merge, 2019-06-08:
sure. x230i isn't supported. look at coreboot for details.

Am 8. Juni 2019 09:51:26 MESZ schrieb Andrew Savchenko <notifications@github.com>:
> After modifying syslinux config to allow greater verbosity, I am
> observing **[1]** the following errors during the boot
>
> ```
> mce: [Hardware Error]: CPU 0: Mahine check: 0 bank 5: ae000000004110a [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

asvc, 2019-06-08:
Thanks for your reply. Would it be possible to incorporate [this path](https://review.coreboot.org/c/coreboot/+/21944/) into the Skulls?   
It adds support for x230i and according to [Jenkins](https://mail.coreboot.org/pipermail/coreboot-gerrit/2017-October/076884.html) shouldn't break things.

I am happy to do whatever testing is necessary.

merge, 2019-06-08:
looks like that would be adding a new laptop to the skulls project. feel free to add it once that is merged.

generally we don't want to add out-of-tree patches.

Am 8. Juni 2019 11:02:29 MESZ schrieb Andrew Savchenko <notifications@github.com>:
> Thanks for your reply. Would it be possible to incorporate [this
> path](https://review.coreboot.org/c/coreboot/+/21944/) into the Skulls?
>
> It adds support for x230i and according to [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

merge, 2019-06-22:
Can we close this issue? Here you tried to run our image on a not-supported Laptop (x230i). We should not confuse people here, but could create a "add support for x230i" issue. ok?

asvc, 2019-06-22:
Apologies for the delay with response. I will swap motherboard with "non-i" version and repeat the test.
Closing for now.

closed 2019-06-22 by asvc