merge/skulls #91: tx and rx pins

Weythran, 2019-06-07:
What does one connect to the tx and rx pins on the raspberri board?

merge, 2019-06-07:
Nothing if you use a keyboard and monitor. Otherwise, this is the serial console to use from your workstation.

How could we improve the documentation? thanks.

Weythran, 2019-06-07:
I'll have a better idea how to improve documentation once i'm through the process. Currently I'm getting the error
"No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically."

merge, 2019-06-10:
That's unrelated to a serial connection to a RPi. That error suggests that you haven't wired up the SPI flash clip correctly.

Weythran, 2019-06-14:
The pins look like they're connected properly to me. 
I'll try flashing using the second hardware method. What do we connect the GND wire on the Pomona clip to?

merge, 2019-06-15:
to GND of the spi flasher.

closed 2021-04-28 by merge