merge/skulls #92: Consider signing releases

asvc, 2019-06-07:
[image]

Perhaps consider:
```gitconfig
[tag]
	gpgsign = true
```

merge, 2019-06-08:
Look at release.sh. Don't I sign already? I might have used a new key though that github hasn't fetched yet...

Am 7. Juni 2019 13:13:31 MESZ schrieb Andrew Savchenko <notifications@github.com>:
>[image]
>
>Perhaps consider:
>```gitconfig
>[tag]
>	gpgsign = true
>```

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

merge, 2019-06-20:
I've updated my key on github...

closed 2019-06-20 by merge