merge/skulls #93: Feature: Support for Thinkpad T420

anjannath, 2019-06-10:
Is there any plans to support the T4xx models?
Also, how hard is it to port coreboot to these models, if provided guidance i am up for the task.

Dongerrs, 2019-07-03:
+1
plus T400, T430, X201, X/T60 (basically all the Thinkpads supported by Coreboot) 

Flashed X230 today and after all the reading I've been doing on the subject over last 2 weeks, it was completely painless and hassle free, using Skulls! Very good project and would speed up adoption of Coreboot if this (skull's deployment scripts) were available for other models.

emacsomancer, 2019-07-03:
I was thinking about suggesting also the X220.

But a more general question: does @merge want these all to be under one umbrella? or would it be better done as model-specific forks? I can imagine headaches for either choice.

merge, 2019-07-04:
we are mostly set up to easily add a new device to our source tree here. It's not hard and no technical problem at all. I personally just don't have a different one than an X230. If I would have a T430, there would already be support for it (resulting in a skulls-t430 release package).

(The **20 are more difficult to support for us than the **30 models that have 2 flash chips)

If you need help in adding support for your **30, but promise to be around and be able to test pull images / flashing procedures, we could team up and I'd plan to "blindly" add support for you to test, as soon as I find time. (it would help if you still have the original lenovo flash backups). so, any volunteer?

Dongerrs, 2019-07-04:
I have:
T430
T420
T400
T61
T60 (with ATI, which I know will be more problematic anyway)
X230 (already running Skulls, tyvm)
X201
X601 (X60 with X61 mobo installed, but still have the X60 mobo that I can test with, as i know the X61 is unsupported)

Both T430/420 are running Quad Core CPUs, but I don't believe this should be an issue, as the upstream Coreboot builds support Quads (these are direct (drop in) replacement CPUs with no H/W mods to make them work). I have read that libreboot has/may have had trouble with quads on the T400 due to the commit those builds are tied to. T400 with quad core has H/W mods to make them work anyway, and it was a specific commit/regression that broke them(?)

Happy to assist with testing, as time/work permits (my partner gets arsey with me if i spend **all** my spare time messing with tech!). I guess the easiest to test first would be the T430, if you are almost ready anyway. Let me know when you have a build ready and I will break it down and give it a go (it's due a rebuild/upgrade of OS anyway, so gives me an excuse to do this!)

Cheers
Dongerrs

merge, 2019-07-04:
do you have a lenovo-bios backup of your T430?

Dongerrs, 2019-07-04:
I dont, but can look to get one if you need it...? Do you know what size SOICS on the 430 tho? I only have 8 pin pomona clip, and im not sure if the 16 pin (cheap) chinese one i have will work....I can try tho...!

merge, 2019-07-04:
ok. let's try without. you should have 2 8-pin chips, just like you do on the X230, see https://doc.coreboot.org/mainboard/lenovo/xx30_series.html is that right?

Dongerrs, 2019-07-04:
Just looking now, after breaking down my t430....gimme a few mins

Dongerrs, 2019-07-04:
OK..Got the backups for you, do you have somewhere secure for me to upload to (I trust you with my mobo's identity, but not the rest of the internet ;) )

8Mb Chip - on left hand side as look at the mobo with the ROM chips towards me and the CPU away from me is a WinBond W25Q64.V
4Mb Chip - on right hand side as you look at the mobo from same position, was identified by flashrom as either:
Macronix MX25L3205(A)
Macronix MX25L3205D/MX25L3208D
Macronix MX25L3206E/MX25L3208E
Macronix MX25L3273E

Tried Macronix MX25L3206E/MX25L3208E first but it didnt find anything so then tried MX25L3205D/MX25L3208D and it read OK (didnt try the other 2)

Cheers
Dongerrs

kraap, 2019-07-04:
Hi guys!
I would be interessted in seeing this project working for T430 model, as weel, and could (try to) help testing some features, if needed!
I also have some backup of my bios (made following this https://nm-projects.de/2017/08/flashing-coreboot-on-the-t430-with-a-raspberry-pi), if you need some moar, just tell me.
My goal would be to deploy and test u-root & heads, as well as beeing able to boot windows - just in case (some acpi tables to review i guess), on t430 model.
So maybe Skulls could be an easier way to do it, and share the good word ? :)

merge, 2019-07-04:
ok. I created an onionshare receive-site temporarily. can you use torbrowser and visit http://onionshare:enjoying-creatable@ypb5sx75z2onvgz5y7nmaap3dwr5nsdtzffblq3qu3yu4zvfcy3bsiad.onion to send files?

Dongerrs, 2019-07-04:
Files should be shared now...
T430_4MB_Factory_Right.rom
T430_8MB_Factory_Left.rom

merge, 2019-07-04:
thanks. I'll have a look during the upcoming days... I hope

Dongerrs, 2019-07-04:
Lol no worries, keep me posted, I'll put the 430 back together now, an wait for your update!
Will have to wait to flash it now, otherwise we wont get a good test. Shame as it is in bits now and i could flash something to it. 

Do you want a copy of the the 420 factory bios too..? Will maybe look to get a backup of that this afternoon, after rebuilding the 430 ;)

Dongerrs, 2019-07-04:
Just thought actually, my 420 BIOS isnt factory bios, its an advanced bios (whitelist removed, advanced menu available, RAM speeds selectable etc), so probably wouldn't make a good backup image to work on

@anjannath - as you started this thread with a request for 420 support, do you have a vanilla/factory 420 BIOS that can be used..?

kraap, 2019-07-04:
Not so much more than a Dell Latitude 6410 (Intel QM57 05) i think, but i'll check soon.
Not sur about the "portability hassle" on this one: the last coreboot supported Dell is the XPS M1530 (northbridge: Intel PM965). 

Idk if many people will be interested (popular laptop..?) but i could dump flash, and start looking at it, in the next days.

Dongerrs, 2019-07-05:
@Merge - Do you have a list/specific command sequence that you use for skulls builds? Thinking about order in which you run things. Something along the lines of:
1 Backup current/factory ROM
2 Run ME Cleaner
2 Extract Blobs from ROM
3 Add blobs to build tree
4 Build coreboot (including extracted blobs)
5 Flash new ROM

Is that documented anywhere...?

merge, 2019-07-08:
@Dongerrs I don't know what you mean. Installing a pre-built binary is documented (as well as running me_cleaner and so on), and rebuilding our image is different story, here: https://github.com/merge/skulls/blob/master/x230/README.md#how-to-reproduce-the-release-images We include everything needed to build.

anjannath, 2019-07-08:
@Dongerrs I do have a T420 with vanilla BIOS, but i am out of station and it'll be a few days before i can make a backup and share it, but will certainly do it once am back

rummik, 2019-08-02:
I have clean BIOS dumps of two T430 ThinkPads -- one of which has an Nvidia GPU

kounsokiri, 2019-09-02:
Hi, i also have a t420 available for testing and have (minor) experience with flashing..

jcholsap, 2019-09-16:
> If you need help in adding support for your **30, but promise to be around and be able to test images / flashing procedures, we could team up and I'd plan to "blindly" add support for you to test, as soon as I find time. (it would help if you still have the original lenovo flash backups). so, any volunteer?

I will ship you a T430s for free and also be available for testing. Could that interest you?

kounsokiri, 2019-09-16:
Yeah im up for that. I don’t have any original flash backups. But I extracted some from install files.

jcholsap, 2019-09-16:
No worries, I'll take care of all that for you. Plus I'll move and socket the EEPROM, just like on the desktop mobos, and give you a few DIP8 chips to make it all easier. I just need a couple of weeks to first finish up another project. How is USPS international shipping for you?

> Yeah im up for that. I don’t have any original flash backups. But I extracted some from install files.

Thrilleratplay, 2019-09-16:
The X220, which is similar to the T420, and the T430 are already part of [Thrilleratplay/coreboot-builder-scripts](https://github.com/Thrilleratplay/coreboot-builder-scripts) and should not be difficult to add to skulls.  The main difference is the T430 and (I think) the T420 both have NVIDIA GPU options.  Fully supporting these models would require a version for Intel and NVIDIA GPUs.

jcholsap, 2019-09-16:
That would be sweet, using the NVIDIA GPU! But questions are already swarming around Pooh's honey pot: It'll boot using the Intel HD GPU but how would you force it to the NVS 5400 to test it? What are your ideas for the power management of using the GPUs? If (probably?) the libfx driver doesn't support NVIDIA, is their an open source alternative available, and is it any good?

Personally I like the idea. And it may just be my ignorance, but I'm wondering about how to define the project parameters.

> The X220, which is similar to the T420, and the T430 are already part of [Thrilleratplay/coreboot-builder-scripts](https://github.com/Thrilleratplay/coreboot-builder-scripts) and should not be [...]

Thrilleratplay, 2019-09-16:
@jcholsap   Sorry, let me clarify.  I support the Intel version of the T430 in coreboot-builder-scripts.   Skulls can support the x230 with a single laptop because the variations in hardware are very small in this model.  When supporting the T420 and T430 with the same support as the x230, with both free and VGA blob configs, there would need to be one of each test machine for the NVIDIA and Intel versions to ensure there are checks in place and working to prevent a user from messing up their machine.  It may seem simple to add additional support to similar models but do not forget that when supporting a user base with a project like this, all of the permutations need to be considered.

jcholsap, 2019-09-16:
Yeah, I looked at your repo. Nice work, BTW. I'll post my reply there to keep this thread clean.

> @jcholsap Sorry, let me clarify. I support the Intel version of the T430 in coreboot-builder-scripts. Skulls can support the x230 with a single laptop because the variations in hardware are very [...]

flawedworld, 2019-09-16:
The ThinkPad T430, T530, W530 and T430s only have TPM support from coreboot 4.9. All of these boards are trivial copy pastes from their **20 counterparts with only needed to adjust from gpu configs and coreboot board names. There is groundwork laid already for a T430, T420, T530 and T520 port from my understanding. We just need heads to be brought up to 4.9 or newer to proceed along with the relevant patches.

jcholsap, 2019-09-16:
I can contribute about (3) T430s +dGPU devices for now but can't commit much brain power until January probably.

merge, 2019-09-17:
> The ThinkPad T430, T530, W530 and T430s only have TPM support from coreboot 4.9. All of these boards are trivial copy pastes from their **20 counterparts with only needed to adjust from gpu configs [...]

that all applies to Heads only, not to Skulls. We don't even use the TPM here.

merge, 2019-09-17:
> I can contribute about (3) T430s +dGPU devices for now but can't commit much brain power until January probably.

ok, I guess I could help supporting the T430s if I had the hardware - most likely not before November, but still..

kounsokiri, 2019-09-17:
> No worries, I'll take care of all that for you. Plus I'll move and socket the EEPROM, just like on the desktop mobos, and give you a few DIP8 chips to make it all easier. I just need a couple of weeks to first finish up another project. How is USPS international shipping for you?

USPS seems fine from what I have read on their site.. How do we proceed from here?    I want to state clearly that I only can help with testing and give feedback  since I’m tinkering about without in-depth knowledge about dev, of which I’m eager to learn ,so please bear with me if I n00b...

jcholsap, 2019-09-17:
Okay, okay. Looks like we have some testers then. I'll post back in one or two weeks. Meanwhile I'll try to have them ready. Also you may want to protect your privacy. My idea is to use a shipping agent, or two, or three... Other ideas are welcomed, of course.

merge, 2019-09-18:
@jcholsap I'd just give you an address via email: martink@posteo.de but since I don't see git commits from you on github, connecting your email to your github user, let's see if I'll know that it's really you :)

jcholsap, 2019-09-18:
I'll take care of that. Just super busy, that's all. Want to keep the thread pertinent. But I'm getting excited about it. Supporting the T430s "shouldn't" be hard. But it never goes that way. It's better to work together. About privacy, I just respect other's rights.

kounsokiri, 2019-09-20:
@jcholsap For my adress --> kounsokiri@protonmail.com

jcholsap, 2019-09-23:
As you can see, I made *some* progress this weekend. This is a project I've been formulating for a year and am finally now making it happen.