merge/skulls #95: please clarify hardware setup & powersupply?

amit1z, 2019-06-15:
First off, thanks for your great project and your talk!
I still have some more questions about the hardware setup:

1. Has anyone had success with the **Raspberry Pi 3 B+** (+ **Pomona 5250**) on **X230T** ? (I've noticed you only write Raspberry Pi 3 in the instructions + the picture seems to be showing a Raspberry Pi 2 B v1.2?).
2. **What powersupply** for the RasPi you use? Is 5V / 1A ok?
3. Did you **disconnect the CMOS battery** on the motherboard either? (in addition to removing the akku, as suggested here [https://0xdd.org/x230coreboot/ ](https://0xdd.org/x230coreboot/): _"be sure to disconnect the main battery and the “CMOS” battery"_)
4. Does skulls adher to the recommended **3.3V limit?** here [https://www.coreboot.org/Board:lenovo/x230](https://www.coreboot.org/Board:lenovo/x230)?
_"Use of an SOIC-8 clip such as Pomona 5250 is recommended. You should ideally power flash using the wake-on-LAN (WoL) feature. If that doesn't work, be extremely careful not to supply more than 3.3V to flash (note that this is dangerous in this case). WoL provides the correct voltage and current in a stable manner, and does not power other parts of the board. Be careful not to connect VCC from your programmer or an external power supply while powering flash using the WoL feature."_
5. (On a side-note: Has anyone hardware-flashed over the latest 2019-04-09 original Lenovo-BIOS ([https://support.lenovo.com/us/en/downloads/ds029187](https://support.lenovo.com/us/en/downloads/ds029187) x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)) yet?)

The reason I'm asking is because of the warnings/reports of fried mainboards, like

[https://doc.coreboot.org/flash_tutorial/ext_power.html](https://doc.coreboot.org/flash_tutorial/ext_power.html)
_"Single board computers (like BeagleBone Black / RPi)
Be careful when connecting a flash chip, especially when using a Pomona test-clip. A short circuit or overcurrent (250mA) causes a brown-out reset, resulting in a reboot of the running operating system (and possible loss of remote shell)"._
or
[https://github.com/merge/skulls/issues/68](https://github.com/merge/skulls/issues/68)
_"already bricked it's x230 by flashing skulls"_

I'm new to this flashing business and want to make certain to get the exact same hardware that has proven to be successful, and take any precautions to replicate it the exact same way, and not destroy my precious tablet. :) 

So much thanks in advance!

amit1z, 2019-06-16:
correction to 5.:
* x230 BIOS 2.75 of 2019-04-17 (G2ETB5WW), EC 1.14 (G2HT35WW): [https://support.lenovo.com/us/en/downloads/ds029187](https://support.lenovo.com/us/en/downloads/ds029187)
and
* **X230t Tablet BIOS 2.73 of 2019-04-18** (GCETB3WW), EC 1.14 (GCHT25WW): [https://support.lenovo.com/us/en/downloads/ds029683](https://support.lenovo.com/us/en/downloads/ds029683)

merge, 2019-06-20:
> First off, thanks for your great project and your talk!
> I still have some more questions about the hardware setup:
>
>     1. Has anyone had success with the **Raspberry Pi 3 B+** (+ **Pomona 5250**) [...]

The picture shows a RPi3, but the pinout should be same for a RPi2 when counting from the left.

> 
>     2. **What powersupply** for the RasPi do you use? Is 5V / 1A ok?

Mine can supply more than 1A but I can't tell how relevant that is, sorry.

> 
>     3. Did you **disconnect the CMOS battery** on the motherboard either? (in addition to removing the accu, as suggested here [https://0xdd.org/x230coreboot/ ](https://0xdd.org/x230coreboot/): _"be sure to disconnect the main battery and the “CMOS” battery"_)

I didn't so far. In case we know that this helps, we could improve our documentation...

> 4. Does skulls adher to the recommended **3.3V limit?** here https://www.coreboot.org/Board:lenovo/x230?
>        _"Use of an SOIC-8 clip such as Pomona 5250 is recommended. You should ideally power [...]

I've never had a problem and we here don't document the WoL powering because it adds much complexity. But I have to say that using a CH341A flasher and a dedicated 3,3 power source is probably the safer option that using a RPi. Should we mention that in the docs?

If we have real evidence that we need to document the WoL powering, feel free to try to add it, as simple as possible...

>     5. (On a side-note: Has anyone hardware-flashed over the latest original Lenovo-BIOS yet?) [edit]
>        x230 BIOS 2.75 of 2019-04-17 (G2ETB5WW), EC 1.14 (G2HT35WW): https://support.lenovo.com/us/en/downloads/ds029187
>        **X230t Tablet BIOS 2.73 of 2019-04-18** (GCETB3WW), EC 1.14 (GCHT25WW): https://support.lenovo.com/us/en/downloads/ds029683
> 

No, but I can't imagine that making any difference.

> The reason I'm asking is because of the warnings/reports of fried mainboards, like
>
> https://doc.coreboot.org/flash_tutorial/ext_power.html
> _"Single board computers (like BeagleBone Black / RPi) Be [...]

I guess you won't be 100% safe, but if you create your backup (using our scripts) and don't rush through it all too much, you should be fine. We've seen many people using our images successfully now.

> 
> So much thanks in advance!

amit4z, 2019-08-25:
Sorry for taking so long. I'm still sifting through all the tutorials i could find, but now have a better conception of how I want to do it (RPi and WOL).

> The picture shows a RPi3, but the pinout should be same for a RPi2 when counting from the left.

Thanks, sorry, looked a bit fuzzy :)

So a 3B+ would work too, right?

> > 2. **What powersupply** for the RasPi do you use? Is 5V / 1A ok?
> Mine can supply more than 1A but I can't tell how relevant that is, sorry.

Sorry for not finding it myself:
_The Raspberry Pi 3 is powered by a +5.1V micro USB supply. Exactly how much current (mA) the Raspberry Pi requires is dependent on what you connect to it. We have found that purchasing a 2.5A power supply from a reputable retailer will provide you with ample power to run your Raspberry Pi._ ([raspberrypi.org](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md))

> I didn't so far. In case we know that this helps, we could improve our documentation...

Another oversight of mine. The user desoldered SPI1 and SPI2. Sorry.
 I could try disconnecing it though, as it lives around the corner anyway.

> I've never had a problem and we here don't document the WoL powering because it adds much complexity. But I have to say that using a CH341A flasher and a dedicated 3,3 power source is probably the safer option that using a RPi. Should we mention that in the docs?

I couldn't find any specifics of how (or if - except your [new report](https://github.com/merge/skulls/issues/97)) people bricked their X230/t because of a RPi.

But there are reports of the [CH341A supplying 5V](https://www.chucknemeth.com/flash-lenovo-x230-coreboot/) ([and here](https://www.onetransistor.eu/2017/08/ch341a-mini-programmer-schematic.html) or [here](https://www.reddit.com/r/coreboot/comments/8b5m8g/ch341a_usb_flasher_and_335v/)) instead of 3.3V - which should not affect skulls, as it recommends the separate AMS1117 power supply, but should be mentioned as a precaution, just in case?
 January 8, 2019: _"As a user commented below, some of the black CH341a programmers may be providing 5V to from GND to the other pins. Eg: GND>MOSI=5V; GND>MISO=5V; Gnd>CLK=5V; Gnd>CS=5V. Sure enough, my programmer measured 5V to the other pins. Luckily no harm was done to my x230 and I flashed it multiple times using the ‘faulty’ programmer. If yours measures 5V to the other pins there are modifications you can do to fix the issue or flash at your own risk."_ ([this](https://www.chucknemeth.com/3-3v-ch341a-modification/) [fix](https://www.eevblog.com/forum/repair/ch341a-serial-memory-programmer-power-supply-fix/), or [here](https://www.onetransistor.eu/2017/08/ch341a-mini-programmer-schematic.html))

There also seem to be many different manufacturers of the CH341A?
 This, and the single purpose use-case of the CH341A vs. the versatility and widespread use and acclaim of the Raspberry Pi, and its cheap 2nd hand availability, made me opt for the later.

Perhaps the Beaglebone Black could be added as a libre option later on?

> If we have real evidence that we need to document the WoL powering, feel free to try to add it, as simple as possible...

As soon as I'm smart enough I could write a detailed step-by-step, "Explain-Like-I'm-5" RPi/WOL-guide from a first-timers perspective, going into all the details I would like to see, which may be obvious to people doing this on a regular basis!

In case of skulls issue tracker's bricking reports like [1](https://github.com/merge/skulls/issues/68), [2](https://github.com/merge/skulls/issues/97), ([3](https://github.com/merge/skulls/issues/49)), I think it would be useful if people were asked / elaborated on their misfortune.
**We should definitely document all bricking instances**, once known in detail, like their steps, PSU, if they ran e.g. thinkpad-ec previously, etc, at a central place like the [x230 coreboot wiki](https://www.coreboot.org/Board:lenovo/x230).
 To prevent possible damages and figure out a foolproof and safe way that lives up to skulls "aim to be the easiest possible coreboot distribution - both to install and to use." credo.

What would definitely help me for now, imo, is if you detailed step-by-step which devices you connect when, in what order, when to power on the RPi, etc!

ami7az, 2019-10-31:
### Users stating the **CH341A's voltage problem also extends to it's other pins:**
_"The chip is powered from 5V, so its **I/O ports will also use 5V**. Basically this is a 5V device. The problem is that any memory you fit in the socket will be powered from 3.3V. And this seems to be the only function of the 3.3V regulator (besides the 3.3V pin on the SPI connector). I don't know why the designer even used a regulator, if it didn't provide a switch to choose between 3.3V or 5V levels and supply."_ [onetransistor.eu: CH341A Mini Programmer Schematic and Drivers](https://www.onetransistor.eu/2017/08/ch341a-mini-programmer-schematic.html)

_"was sending **5V to the signal outputs** instead of 3.3V)"_ [chucknemeth.com: 3.3V CH341a Signal Output Modification](https://www.chucknemeth.com/3-3v-ch341a-modification/)

_"**all I2C and SPI signals are 5V** while the external memory is powered at 3.3V."_ [eevblog.com: CH341A Serial Memory Programmer Power Supply Fix](https://www.eevblog.com/forum/repair/ch341a-serial-memory-programmer-power-supply-fix/msg1323775/)

_"From 3.3V pin to ground, the actual voltage was about 5.5V. The **same for other pins (MOSI, MISO etc). From 5V to ground, the actual voltage was 9.5V.**"_
??? _"newer versions of CH341A boards that allow you to set the voltage specifically at 3.3V. Here are two versions:"_ [reddit.com: CH341A USB flasher and 3.3-5V](https://www.reddit.com/r/coreboot/comments/8b5m8g/ch341a_usb_flasher_and_335v/)

(_"black pcb version supplies 5V to the chip"_ [win-raid.com: [Guide] Using CH341A-based programmer to flash SPI EEPROM](https://www.win-raid.com/t796f16-Guide-Using-CH-A-based-programmer-to-flash-SPI-EEPROM-5.html))

### 
As this seems to be a common CH341A problem (even though from just skimming through it users might still come out lucky), and as it occurs independent of skulls' recommended AMS1117 PSU, and seems finicky to fix, and there are enough vague bricking reports out there, I would definitely **investigate this issue further, and put a warning in [Readme.md](https://github.com/merge/skulls/blob/master/x230/README.md)?**.
(Sorry for missing this the first time around, despite even quoting it.)