merge/skulls #96: query: 'moving' from a coreboot'ed & me_clean'ed

emacsomancer, 2019-06-19:
A couple of years ago, I installed  `me_clean`ed version of coreboot on my x230, using the `-S` option for `me_cleaner` ([it looks like `skulls` does the same thing](https://github.com/merge/skulls/blob/master/x230/external_install_bottom.sh#L230)). 

What is the procedure if I want to move to skulls? Would it be as simple as running `./x230_skulls.sh`?

(And does this save a backup of the current firmware somewhere? Or should that be done separately/manually?)

merge, 2019-06-20:
> A couple of years ago, I installed `me_clean`ed version of coreboot on my x230, using the `-S` option for `me_cleaner` ([it looks like `skulls` does the same thing](https://github.com/merge/skulls/blob/master/x230/external_install_bottom.sh#L230)).

applying me_cleaner and flashing a coreboot bios image should be seen as 2 different things, at least on the X230. You need to flash externally to do that, but since you already have, there's  nothing to do for you. (we apply `me_cleaner -d -S` now but the difference is not much.)

> 
> What is the procedure if I want to move to skulls? Would it be as simple as running `./x230_skulls.sh`?

yes, if you have unlocked the IFD back when you flashed externally the first time, this will just work.

> 
> (And does this save a backup of the current firmware somewhere? Or should that be done separately/manually?)

no. we only save a backup in our "external" scripts. If you have a hardware flasher, I'd suggest you run  our external scripts first. you don't have to though. you probably don't need a backup of your current coreboot system.

emacsomancer, 2019-06-20:
> applying me_cleaner and flashing a coreboot bios image should be seen as 2 different things, at least on the X230. You need to flash externally to do that, but since you already have, there's nothing to do for you. (we apply me_cleaner -d -S now but the difference is not much.)

Right, reviewing what this process looked/looks like, it's really two separate flashing events.

> yes, if you have unlocked the IFD back when you flashed externally the first time, this will just work.
Thanks. Yep, booting with `iomem=relaxed` and running `.x230_skulls.sh` as root/sudo worked perfectly fine! Maybe a brief section 'Moving from other coreboot installations' which mentions this? 

> no. we only save a backup in our "external" scripts. If you have a hardware flasher, I'd suggest you run our external scripts first. you don't have to though. you probably don't need a backup of your current coreboot system.

I saved a backup of my current firmware manually just in case. It produced a 12M file which I assume would need to be split and externally flashed in case of issues. They weren't any, so it wasn't needed.

It's nice to have a quick and easy way of updating coreboot - I hadn't updated coreboot since I first externally flashed it. Thanks for this project!

closed 2019-06-20 by emacsomancer