merge/skulls #99: [Discussion] x220 ThinkPad | Laptop Support Request

WorryTheBirds12, 2019-08-01:
**Discussion**: Running Skulls on the [ThinkPad x220](https://www.thinkwiki.org/wiki/Category:x220)
**Why?** The x220 is the only decently built modern mechanical keyboard laptop that can run a modern OS. To many users this is a big benefit. 

_______________________

Want to support this issue? Post a [bounty on it]()! We accept bounties via [BountySource](https://www.wikipedia.org/wiki/Bountysource).

[image]

emacsomancer, 2019-08-01:
(Cf. https://www.thinkwiki.org/wiki/Install_Classic_Keyboard_on_xx30_Series_ThinkPads )

abhas, 2019-08-01:
I have the x220 keyboard running on my x230 with coreboot/skulls on it. One of the x230 units had coreboot on it already and so the x220 keyboard behaves like the x230 keyboard would (with some mis-mapped and dead keys). On another one I patched the ec firmware before flashing skulls on it and hence the keyboard behaves exactly like it is supposed to.

So while getting skulls to work on the x220 would be useful for a variety of reasons, if you just want the x220 keyboard then you could do that easily with the x230 to. The link posted by @emacsomancer above has all the details.

elagost, 2019-09-23:
I have an x220 running coreboot (not skulls specifically, but just compiled it myself) that I would be willing to test with. I just flashed an updated build today, so it's still working with recent builds.
I think the biggest issue is that there are 2 types of flash chips from what I understand (Macronix MX25L6405, or Winbond W25Q64.V) but Coreboot detects both. Anyone else have a testing machine?

merge, 2019-09-24:
for the x220, the only challange is writing good "external_" scripts for bootstapping, that use the machine's ME, gbed and so on. We want the user to only have to connect with an external clip _once_. But if you have an x220 and enough time, it's no problem

elagost, 2019-10-01:
Slowly getting around to looking at this. I've noticed your configs (nonfree-defconfig-* and free-defconfig-* in the x230 folder) are really small but mine when building coreboot for my x220 ended up much larger.
My config is here: https://gist.github.com/elagost/b323e9ca089719426b7761f077bebd43#gistcomment-3042606

Any reason this would be the case? Am I over-complicating things?

Thrilleratplay, 2019-10-02:
@elagost   Those are [defconfigs](https://stackoverflow.com/a/41886394), they are a stripped down version of the generated config file.  If you run `make savedefconfig` you will get the stripped down version too.  It is to make migrating to newer versions easier.

elagost, 2019-10-02:
Oh, much easier! This is mine for my x220's current setup:
```
CONFIG_USE_OPTION_TABLE=y
CONFIG_USE_BLOBS=y
CONFIG_VENDOR_LENOVO=y
CONFIG_HAVE_IFD_BIN=y
CONFIG_BOARD_LENOVO_X220=y
CONFIG_DRIVERS_PS2_KEYBOARD=y
CONFIG_CONSOLE_POST=y
CONFIG_NATIVE_RAMINIT_IGNORE_MAX_MEM_FUSES=y
CONFIG_SEABIOS_PS2_TIMEOUT=3000
CONFIG_H8_SUPPORT_BT_ON_WIFI=y
CONFIG_HAVE_ME_BIN=y
CONFIG_USE_ME_CLEANER=y
CONFIG_HAVE_GBE_BIN=y
CONFIG_SEABIOS_THREAD_OPTIONROMS=y
CONFIG_COREINFO_SECONDARY_PAYLOAD=y
CONFIG_NVRAMCUI_SECONDARY_PAYLOAD=y
```

Thrilleratplay, 2019-10-02:
It looks [very similar to the one I use](https://github.com/Thrilleratplay/coreboot-builder-scripts/blob/master/x220/defconfig-4.10).  Remember that is just a defconfig save it as defconfig on the coreboot directory and do `make defconfig` to repopulate your config.  Take a look at [common/config_and_make.sh](https://github.com/merge/skulls/blob/master/common/config_and_make.sh) to see the commands skulls uses.

valner88, 2020-02-01:
> I have the x220 keyboard running on my x230 with coreboot/skulls on it. One of the x230 units had coreboot on it already and so the x220 keyboard behaves like the x230 keyboard would (with some [...]

I could not manage to boot the dos usb flasher, on my x230t. It stays loading indefinitely, i cannot also flash my bios using winflash or similar, it could be a tell of the problem. The usb flasher was done correctly using the correspondent methods hamish coleman. Any ideas? I want to use my x220 keyboard and so far all i have got is a downgrade.