merge/skulls #100: Adding sha256sum integrity checks (+ hashes) for backup roms? (And ROM-backup general)

amit4z, 2019-08-25:
Adding sha256sum integrity checks (+ hashes) for backup roms? (And ROM-backup general)

- 1
Could you please also **add a sha256sum compare + hash-files for the backup ROMs** after 
> [_cmp --silent "${TEMP_DIR}"/test1.rom "${TEMP_DIR}"/test2.rom_](https://github.com/merge/skulls/blob/master/x230/external_install_bottom.sh)

 like you already do for the skulls ROMs, e.g.:
_sha256sum x230-top-rom-"${CHIPNAME}"-backup-timeanddate.rom > x230-top-rom-"${CHIPNAME}"-backup-timeanddate.rom.sha256_
 This **could ensure the integrity of the backup-files** at a later date!
 (On that note, would it be possible to immediately add an integrity check, perhaps in RAM or through pipe or similar, so the backup can't get corrupted as it's being saved on a broken sdcard? Or is this overly paranoid? (I just lost files in the past because of bad flash/SD-card-storage).)

- 2.1
Also, am I missing something or are the scripts missing an error check in case of an integrity mismatch? (Sorry in advance, I'm not a good programmer)
[external_install_top.sh](https://github.com/merge/skulls/blob/master/x230/external_install_top.sh) and [external_install_bottom.sh](https://github.com/merge/skulls/blob/master/x230/external_install_bottom.sh) only seem to use
> _wc -c <"$TEMP_DIR/test1.rom"_

to compare the first readout to a preset _reference_size_ byte-size, and exit if this differs, but the cmp between test1.rom and test2.rom is missing an exit code or follow-up?

- 2.2
In [x230_skulls.sh](https://github.com/merge/skulls/blob/master/x230/x230_skulls.sh) sha256 is only tested if a new update is curled, not for the supplied roms and their hashes?

- 3.1
Would it be possible for skulls to make backups of both chips first (preferrably internally) with flashrom, before hardware-flashing?
 This way the user could secure it e.g. on a thumbdrive before getting started...

- 3.2.
In case of a crash: at the moment I have no way to secure the backups to an external device outside the RPi or 2nd PC, without disturbing the flash-procedure that follows immediately afterwards?
 Maybe a _sync_ before flashing would be useful, and maybe have a pause/keyboard-input to continue (so the user can make a backup manually before getting started)? (but 3.1 would probably be better)

- 4
Could there be **generic backup roms**, e.g. extracted or made from the official Lenovo BIOS update iso ([x230](https://support.lenovo.com/us/en/downloads/ds029187) / [x230t Tablet](https://support.lenovo.com/us/en/downloads/ds029683)), as a fallback?

- 5
As long as the hardware is not broken, e.g. by overcurrent: If I have the original ROMs backuped, can I always in-system-flash them back, no matter what the current content of the chips is (e.g. independent of where the last flash aborted or messed up, brown-out-reset, etc), right?

Thanks so much for your project! The more I read about all this the more I realize how many things have to be considered, and how much work this actually is. Having everything condensed in one small, easy script and package is just amazing!

merge, 2019-08-26:
> Adding sha256sum integrity checks (+ hashes) for backup roms? (And ROM-backup general)
>
>     * 1
>       Could you please also **add a sha256sum compare + hash-files for the backup ROMs** after

sure, why not.

> like you already do for the skulls ROMs, e.g.:
> _sha256sum x230-top-rom-"${CHIPNAME}"-backup-timeanddate.rom > x230-top-rom-"${CHIPNAME}"-backup-timeanddate.rom.sha256_
> This **could ensure the [...]

true. you should be able to flash any image you want with these scripts. That's why we only check for the correct size.

> 
>     * 2.2
>       In [x230_skulls.sh](https://github.com/merge/skulls/blob/master/x230/x230_skulls.sh) sha256 is only tested if a new update is curled, not for the supplied roms and their hashes?

again, you should not only be able to flash the included roms, but any rom you build or a newer one we release. you can easily check the file-integrity of a rom manually, if you want to.

> 
>     * 3.1
>       Would it be possible for skulls to make backups of both chips first (preferrably internally) with flashrom, before hardware-flashing?
>       This way the user could secure it e.g. on a thumbdrive before getting started...

don't we do that using the `-k` option?

> * 3.2.
>       In case of a crash: at the moment I have no way to secure the backups to an external device outside the RPi or 2nd PC, without disturbing the flash-procedure that follows immediately [...]

I'm not sure what you mean.

> 
>     * 4
>       Could there be **generic backup roms**, e.g. extracted or made from the official Lenovo BIOS update iso ([x230](https://support.lenovo.com/us/en/downloads/ds029187) / [x230t Tablet](https://support.lenovo.com/us/en/downloads/ds029683)), as a fallback?

a "generic backup rom" from our point of view would just be a released skulls image. For flashing back a Lenovo bios, use your backups and flash them externally. Flashing them internally would need be to tested. Would be interesting to see if that's possible, flashing from zero to end, without a layout file.

> 
>     * 5
>       As long as the hardware is not broken, e.g. by overcurrent: If I have the original ROMs backuped, can I always in-system-flash them back, no matter what the current content of the chips is (e.g. independent of where the last flash aborted or messed up, brown-out-reset, etc), right?
> 

I'm not sure. Disassemble and flash externally to be safe.

> 
> Thanks so much for your project! The more I read about all this the more I realize how many things have to be considered, and how much work this actually is. Having everything condensed in one small, easy script and package is just amazing!

amit4z, 2019-09-15:
>> Could you please also add a sha256sum compare + hash-files for the backup ROMs

> sure, why not.

thanks so much!

>> compare the first readout to a preset reference_size byte-size, and exit if this differs, but the cmp between test1.rom and test2.rom is missing an exit code or follow-up?

> true. you should be able to flash any image you want with these scripts. That's why we only check for the correct size.

Got it. It would be amazing if there could be an easy [build.sh](https://github.com/merge/skulls/blob/master/build.sh)-way to compile your own skulls-ROM with [truncated IME](https://github.com/merge/skulls/issues/101) someday!

>>  Would it be possible for skulls to make backups of both chips first

> don't we do that using the -k option?

>> without disturbing the flash-procedure

> I'm not sure what you mean.

My fear is that my computer could crash during the flash-procedure, and take the new backups with it (currently a _sync_ and/or _read_ (pause) command between backup and flashing is missing?).
Or that a corrupt micro-sdcard in the RPi could render the backup-ROMs unusable. 
I just wanted to avoid rummaging about with the RPi while the whole operation is in progress...

> a "generic backup rom" from our point of view would just be a released skulls image.

Got it. I didn't think as much as doing an official Lenovo-USB-flash, once a botched flash-attempt has been restored externally with a skulls ROM.

> For flashing back a Lenovo bios, use your backups and flash them externally. Flashing them internally would need be to tested. Would be interesting to see if that's possible, flashing from zero to end, without a layout file.

Yes, would be great to see if that worked!!

>> As long as the hardware is not broken

> I'm not sure. Disassemble and flash externally to be safe.

This I meant by in-system-flashing, sorry. ([In-system programming](https://en.wikipedia.org/wiki/In-system_programming))

Thank you!