merge/skulls #103: which E-Z Hook?

amit4z, 2019-09-15:
Could you please link/name which [E-Z Hook](https://geizhals.eu/?fs=e-z+hook) model is the right one? There are so many...
Thanks!

Thrilleratplay, 2019-09-16:
I use [E-Z Hook XKM](https://catalog.e-z-hook.com/item/test-hooks/e-z-micro-hooks-single-hook-style/xkm-1).

amit4z, 2019-10-12:
Thanks @Thrilleratplay! In case you're using a RPi, do you have any particular order in which you're connecting the hooks to the chip, to the RPi and when you're powering on the latter?

Thrilleratplay, 2019-10-12:
@amit4z  To save you from the trial and error, you'll find out that it is very easy to get the probe hooks to bridge between SOIC pins when putting them on so you will want to put the hooks on the chip first before there is any power going through them.  Also,until you be come comfortable with the hooks double check that you have not hooked two pins at once.  It isn't easy if you are paying attention to do this but I've managed a few times.  Once the hooks are on, they are not going anywhere.  The pins on the probe header are smaller than the GPIO pins on an rPi and well easily slide out.  This can be a little annoying.  I can't find the post on the eevblog where someone replaced the female jumper wire end with a wire crimp.  I put some heat shrink wrap around the end, heated it and crammed it in pin slot so it wouldn't slide out.  Electrical tape would be the same.  Initially, just use small pieces scotch tape to hold the female jumper wire end the probe.

ami7az, 2019-10-31:
Thank you, this was very informative! I found a site which has some [pictures/comparisons](https://sigrok.org/wiki/Probe_comparison) of them and similar others like Pomona Grabbers in action. It looks rather delicate, but on the other hand they might be more universal if one is dexterous enough.

Would this work?
1. Tape female jumper-wire connectors to ends of EZ-Hook XKM probes to prevent them from sliding off.
2. With removed x230 battery and removed PSU, and RPi-side wire-ends unplugged, connect each hook to its required **bottom BIOS-chip** pin (see [Readme.md](https://github.com/merge/skulls/blob/master/x230/README.md)).
3. Make absolutely certain **no hook connects/shortens two pins** at once.
4. On powered off/PSU-unplugged RPi: connect opposing jumper-wire-ends.
5. **If flashing with WOL** (Wake On LAN), make sure the power-jumper-wire+hook: **(VCC 8) is not connected**, plug x230 PSU, then plug active LAN-cable. Ensure x230's LAN-port light goes on, and x230 remains off.
6. Double-check everything! :)
7. Power/plug on RPi, and proceed with skulls' [external_install_bottom.sh](https://github.com/merge/skulls/blob/master/x230/external_install_bottom.sh)
8. Once done, unplug PSU from x230, remove LAN-cable, shut down RPi, unplug its PSU, disconnect RPi-side jumper-wire-ends, then the hooks off the BIOS-chip.
9. Repeat 1. - 8. with **top BIOS-chip** and [external_install_top.sh](https://github.com/merge/skulls/blob/master/x230/external_install_top.sh).

(Maybe our thread is also of use to [#115](https://github.com/merge/skulls/issues/115))

Do you by any chance also know if I have to do this hardware-flash-procedure again, if I want to shrink/move IME later on, or if software-flashing for it will be enough, once the IFD is unlocked here?

Thrilleratplay, 2019-10-31:
@ami7az I do not use the wake on lan method when flashing, I use the v3.3 power from the programmer however I can confirm  the rest is correct.  In theory, yes you should be able to shrink the ME size internally afterwards however there are very few reasons to do so other than the novelty of saying you did.  The space it too small for a linux kernel and coreboot with all of the secondary payloads still fits in the original partition.

ami7az, 2019-10-31:
Thank you! I want to try it w/ WOL (wich is [recommended by coreboot](https://www.coreboot.org/Board:lenovo/x230) ([also #95](https://github.com/merge/skulls/issues/95)))

GND is still needed w/ WOL?

### 
ME: You could also add free boot-images/coreboot-payloads like Memtest86+. Is there a risk in applying me_cleaner fully? > could you please take a look at [moving IME #101](https://github.com/merge/skulls/issues/101) sometime?

Thrilleratplay, 2019-11-04:
@amit4z  I cannot provide any information for the WOL method.  Resizing ME I think reclaims 1.5M which is too small for OpenBSD or FreeDOS.  This maybe enough for a floppy image payload but SeaBIOS needs a patch implemented first.

ami7az, 2019-11-13:
o.k. thanks!

closed 2020-06-30 by merge