merge/skulls #104: vbios run twice

pgera, 2019-09-15:
Based on some experiments and discussions on irc, it looks like CONFIG_VGA_ROM_RUN=y
is redundant as it causes coreboot to load the rom and seabios loads it again. This also means that CONFIG_PCI_OPTION_ROM_RUN_YABEL=y is not effective if seabios runs the rom later on bare metal. The following options are possible:

- Coreboot doesn't do gfx init, only seabios does it from the vbios
- "in theory you could run the full VBIOS in Yabel, and then let SeaBIOS run the shim VBIOS that is also used for native gfx init" - icon[m]

merge, 2019-09-17:
how does a fix look like? this may be a coreboot Kconfig issue then?

pgera, 2019-09-17:
For the first solution, you select "Graphics initialization (None)" and unselect "Add a Video Bios Table (VBT) binary to CBFS" in the menu since seabios can use the vga bios to initialize the graphics. This is what my defconfig looks like:

```
CONFIG_USE_OPTION_TABLE=y                                                       
CONFIG_BOOTSPLASH_IMAGE=y                                                       
CONFIG_VENDOR_LENOVO=y                                                          
CONFIG_CBFS_SIZE=0x400000                                                       
CONFIG_VGA_BIOS_ID="8086,0166"                                                  
CONFIG_VGA_BIOS=y                                                               
CONFIG_VGA_BIOS_FILE="pci8086,0166.rom"                                         
CONFIG_HAVE_IFD_BIN=y                                                           
CONFIG_BOARD_LENOVO_T530=y                                                      
CONFIG_DRIVERS_PS2_KEYBOARD=y                                                   
CONFIG_PCIEXP_L1_SUB_STATE=y                                                    
CONFIG_PCIEXP_CLK_PM=y                                                          
CONFIG_SEABIOS_PS2_TIMEOUT=10                                                   
CONFIG_H8_SUPPORT_BT_ON_WIFI=y                                                  
CONFIG_HAVE_ME_BIN=y                                                            
CONFIG_CHECK_ME=y                                                               
CONFIG_USE_ME_CLEANER=y                                                         
CONFIG_HAVE_GBE_BIN=y                                                           
# CONFIG_INTEL_GMA_ADD_VBT is not set                                           
CONFIG_COREINFO_SECONDARY_PAYLOAD=y                                             
CONFIG_NVRAMCUI_SECONDARY_PAYLOAD=y                                             
CONFIG_TINT_SECONDARY_PAYLOAD=y
```