merge/skulls #105: Please clarify of flashing X230 (heads,bricks, intel ME cleaner, etc)

gr01d, 2019-09-30:
I am looking forward on getting a X230, I have read issues: https://github.com/merge/skulls/issues/95 , https://github.com/merge/skulls/issues/49 
they both claim that flashing externally is dangerous and may risk of bricking the EC (embedded controller)? I plan to flash it with a modified CH341a (3.3V). I have some questions:

- after flashing this, will this disable intel ME so I wont have to use me_cleaner?
- If after cleaning ME, the gigabit Ethernet port will be useless/disabled?
- does internal flashing work? conflicting information
- how do you make it enter Wake On LAN?
- should you remove cmos battery (CR2032 yellow) before flashing?
- should you plug adapter while flashing?
- does eGPU works with Windows and GNU+Linux after skull-ing

- do you plan to support X220/X220T? 

Please answer them completely.
Thank You,

Thrilleratplay, 2019-09-30:
1. If `external install bottom.sh` run with the `-m` flag on the , the [me_cleaner script is being used](https://github.com/merge/skulls/blob/master/x230/external_install_bottom.sh#L231).
2. Cleaning ME should have no noticeable effect on the machine whether using Skulls or the stock BIOS.  I have seen ME related errors in my dmesg log however everything still works afterwards. Also, the BIOS rom has different partitions that the `ifdtool` can create a layout of which would show the ME and the gigabyte ethernet firmware (usually abbreviated GBE) are separate partitions.
3.  Yes/No.  Does that clarify?  It isn't quiet conflicting, the stock BIOS is locked to prevent malicious attacks.  To get around this, initially, on an X230 you will need to flash externally to get around this lock.  Afterwards, assuming the IFD is unlocked, which is also done in `external_install_bottom.sh`, it will allow you to update using internal flashing method.  So you can flash internally only after you flash externally to get around the lock.
4.  That one I am unsure on.
5.  It is probably a good idea however I have never run into an issue with leaving it plugged in.
6.  The power adapter to the laptop?  No.  There should be no power going to the laptop if using a CH341A.  There are flashing methods that utilize power from the mainboard for the v3.3, however there are risk associated with it and not something I would recommend.
7.  The x230 only has on-board Intel graphics.  Depending on the application, you can use the open source libgfxinit libraries or the Intel blob.  Linux does not have an issue with either or them.  I have had difficulty with correct resolution Windows 7 if using the open source library but it seems to work fine in Windows 10.  Do no install the Intel Drivers without the blob as it crash windows.  There may be some Intel specific configuration or functionality that are lost using Windows and libgfxinit however the main reason users op for including the blob is to have a splash screen on boot.
8. https://github.com/merge/skulls/issues/99

gr01d, 2019-10-01:
Thanks for the quick response, regarding

> The x230 only has on-board Intel graphics. Depending on the application, you can use the open source libgfxinit libraries or the Intel blob. Linux does not have an issue with either or them. I have [...]

Does 3D acceleration works after doing coreboot/heads? I want to try to play some low end games that needs 3D acceleration and maybe game development with unity3d/godot and modelling with blender? 

For the eGPU, I mean adding an external GPU via Expresscard using EXP GDC pci-e adapter to expresscard, will this work?

Thrilleratplay, 2019-10-01:
@gr01d  I do not know the answers to these but I suspect, regardless of BIOS, that the hardware 3D acceleration performance on a Intel HD 4000 was poor when it first came out in 2012.  A x230 is not a good base for 3d graphics.  A Thinkpad T430, the same generation as the x230, had an option NVidia GPU built in which may be more suited for your needs.  However the T430 is not supported by Skulls and disassembling them to access the BIOS chips to flash is a massive pain due to the entire computer needing to be dismantled and the mainboard removed from the chassis.  An external GPU may work however, the expresscard bus would limit speed.

jcholsap, 2019-10-01:
Here's just an idea
https://www.reddit.com/r/thinkpad/comments/9u739u/t430_egpu_step_by_step/

jcholsap, 2019-10-01:
EXP GDC Beast Expresscard mentioned is available on aliexpress.com

Hermanio, 2019-10-13:
I can confirm that at least on Windows 10 the EXP GDC Beast ExpressCard34 will **_NOT_** work if you have Skulls installed. It will start booting, but it will BSOD with ACPI_BIOS_ERROR. Not much luck under Linux, either.

I have had success with a T420 with the same setup and Coreboot installed. This seems to be due to the T420 configuration taking into account the dGPU models, which is why it probably allowed it to work on my Intel HD based T420. I'm not sure how coreboot configuration works exactly, but it could be a possibility that the X230 config could be modified to achieve the same result.

It does work with the stock BIOS, have personally had success with X230 + AMD RX 570.

gr01d, 2019-10-22:
Thanks for your reponses, Unfortunately I got a X200T instead of X230.
For the eGPU part, I guess not flashing with coreboot but just neutralizing the intel ME will make eGPU work?
I will now close this issue.

closed 2019-10-22 by gr01d