merge/skulls #106: Is it worth updating Embedded Controller (EC) to the new version 1.5?

apraile, 2019-10-09:
There is a new official version of [BIOS](https://support.lenovo.com/ch/en/downloads/ds029187) (**2.77**) for X230 that fixes a [vulnerability](https://download.lenovo.com/pccbbs/mobiles/g2uj33uc.txt) in the Embedded Controller (updated to **1.15**). I would like to know if updating the EC has any benefit in terms of security for skulls/coreboot users.

_Note_: the new EC v1.15 is not compatible with [thinkpad-ec](https://github.com/hamishcoleman/thinkpad-ec#compatibilty-warning) patch.

Thrilleratplay, 2019-10-09:
@apraile Security is a relative term.  If you follow the [link to the Lenovo description of the vulnerability](https://support.lenovo.com/us/en/solutions/len-27764) is states 

> could allow a user with administrative privileges or physical access the ability to update the Embedded Controller with unsigned firmware.

 My interpretation of this is it could allow coreboot to modify the EC for better control of a system.  Personally, I would skip this update, however, it is up to you.

jcholsap, 2019-10-09:
So is there no data store within the EC? Are all the ME data files in the EEPROM partitions? Or is he using the ME, having not ran the cleaner?

apraile, 2019-10-09:
@jcholsap I think the _me_cleaner_ tool doesn't touch Embedded Controller only the BIOS. Thanks.

@Thrilleratplay OK, thanks. Considering my threat model (along with my laziness to flash externally :wink:) I don't think I'll update.

Thrilleratplay, 2019-10-09:
@jcholsap  "no data store", not in the way I believe you are thinking.  The EC is separate from the BIOS and deals with lower level controls like translating keystrokes, LCD backlight and locking out third party batteries.  You would probably enjoy [this write up of removing the battery lock out on the x230](http://zmatt.net/unlocking-my-lenovo-laptop-part-1/).  Lenovo's official BIOS update will update the EC and the BIOS firmware (ifd, me, gbe, bios)  but flashrom only can modify BIOS firmware (ifd, me, gbe, bios).  

@apraile And still champion.....laziness.  
[image]

jcholsap, 2019-10-09:
@Thrilleratplay According to this article below, the ME modules are stored in a separate EEPROM.
https://hardenedlinux.github.io/firmware/2016/11/17/neutralize_ME_firmware_on_sandybridge_and_ivybridge.html

Thrilleratplay, 2019-10-09:
@jcholsap The ME firmware is in a separate partition of the same boot rom as the gigabit ethernet firmware (GBE), BIOS (stock or coreboot) and the intel file descriptor (IFD). On the x230 this all two physical SOIC chips, one 8mb and one 4mb. However, ME has its tentacles in other portions of the machine to prevent tampering;  Intel calls this a safety feature but to us it is a pain in the ass that prevents  ME from being completely removed.  [ME cleaner](https://github.com/corna/me_cleaner) leave just enough to trick the system into completing the boot process but not enough to function.

jcholsap, 2019-10-09:
@Thrilleratplay It looks like if the EEPROM is sufficiently powered, then flashrom can write me_cleaner. From the article quoted before:
"Unfortunately, on most platforms with ME, like the example above, the ME region is usually readable only for ME hardware, not the main CPU, which prevents us from using flashrom(8) with internal programmer to even read the whole content of the vendor firmware. In order to research the boot firmware, we need an external programmer."

Thrilleratplay, 2019-10-09:
@jcholsap Yep.  This is why you need to externally flash an x230 the first time and unlock the ifd

jcholsap, 2019-10-09:
@Thrilleratplay Hey, thanks, this helps me prep for my first run at this. I've not done it yet. And thanks for "Dave". So true!!! BTW, I pulled out my T430 machine and double checked. It has what we need for testing dGPU. Getting there...

jcholsap, 2019-10-09:
Yeah, i had come across Dr Matthew Chapman's article before and saved it to my folder. Very interesting because i hadn't thought the batteries were so "smart", just an logic array at most. But it's not so surprising b/c my power tool has smart batteries and for good reason. But for the OEM to authenticate it is a bit intrusive. And it demonstrates the OEM's attitude about "their" products. And I will stop there. Because that's just the start of the rabbit hole.

jcholsap, 2019-10-09:
@apraile This guy here warns against updating:
https://github.com/hamishcoleman/thinkpad-ec/blob/master/README.md#compatibilty-warning

apraile, 2019-10-10:
> @apraile This guy here warns against updating:
> https://github.com/hamishcoleman/thinkpad-ec/blob/master/README.md#compatibilty-warning

Yeah, I've read it [before](https://github.com/merge/skulls/issues/106#issue-504670245). I also use [TLP](https://linrunner.de/en/tlp/tlp.html) to set charge thresholds that [write those thresholds in EC](https://linrunner.de/en/tlp/docs/tlp-faq.html#thresholds-powered-off) and I don't know if EC v1.15 is compatible. At the moment I'm staying with EC v1.14.